espressif/ESP8266_RTOS_SDK #292: 关于MQTT的二次重连问题，概率较高，影响二次开发。

xuhongv, 2018-08-27:
- 关于在 ```MQTT``` 在 ```rtos```的使用，[修复会导致wifi断开的问题](https://github.com/espressif/ESP8266_RTOS_SDK/issues/285) ，但是在查遍了```issue```里面所有关于 ```MQTT``` 重连问题，并没有发现很好的解决方案，于是，我在试着重连的代码之后。重连的成功率还是蛮低的，我的思路是：
  
 > 1、 第一次连接 ```MQTT``` 服务器之后，开启定时器不断轮询是否服务器是否断开？如果是，则发送重连的消息。
>  2、收到重启信息，则先断开底层的 ```TCP``` ，再一次连接服务器，再订阅主题！

上面的步骤，成功率还是很高的！下面是一些代码截图：

-----------------------------------------------------------------------------------

[x] 连接 ```MQTT``` 服务器的核心逻辑代码：

```
static void Task_MqttClient_Connect(void *pvParameters) {

        bool isNeedQueue = true;
	Network network;
	unsigned char sendbuf[2048], readbuf[2048] = { 0 };
	int rc = 0, count = 0;
	MQTTPacket_connectData connectData = MQTTPacket_connectData_initializer;
	pvParameters = 0;
	NetworkInit(&network);
	MQTTClientInit(&client, &network, 5000, sendbuf, sizeof(sendbuf), readbuf,
			sizeof(readbuf));

	bool first_flag = false;

	char *address = MQTT_SERVICE;
	connectData.MQTTVersion = 3;
	connectData.clientID.cstring = originTopic;
	connectData.username.cstring = MQTT_USER_NAME;
	connectData.password.cstring = MQTT_USER_PAW;
	connectData.keepAliveInterval = 30;
	connectData.cleansession = true;

	for (;;) {

		while (wifi_station_get_connect_status() != STATION_GOT_IP) {
			vTaskDelay(1000 / portTICK_RATE_MS);
		}

		printf("Task_MqttClient_Connect NetworkConnect init...\n");

		if ((rc = NetworkConnect(&network, address, MQTT_PORT)) != 0) {
			printf("MClouds NetworkConnect connect fail , code %d\n", rc);
		} else {
			printf(" NetworkConnect init succeed...\n");
		}

                //如果第一次开启了此任务，则下次重连不需要开启了
		if (!first_flag)
			if ((rc = MQTTStartTask(&client)) != pdPASS) {
					printf("Return code from start tasks fail , errorCpode: %d\n", rc);
			} else {
				printf("Use MQTTStartTask succeed...\n");
				first_flag = true;
			}

		if ((rc = MQTTConnect(&client, &connectData)) != 0) {
			printf("[SY] MClouds connect fail , errorCode: %d\n", rc);
			network.disconnect(&network);
			vTaskDelay(1000 / portTICK_RATE_MS);
		}

		if ((rc = MQTTSubscribe(&client, subTopic, QOS0, MessageArrived))!= 0) {
			printf("[SY] MClouds Subscribe fail is %d\n", rc);
			network.disconnect(&network);
			vTaskDelay(1000 / portTICK_RATE_MS);
		} else {
			isConnectMQTT = true;
			printf("MQTT subscribe to topic -> %s\n", subTopic);

		}

		xQueueReset(MqttMessageQueueHandler);

		while (1) {

			char payload[2048];

			struct esp_mqtt_msg_type *pMsg;
			printf("MqttMessageQueueHandler waitting ..\n");

			xQueueReceive(MqttMessageQueueHandler, &pMsg, portMAX_DELAY);

			if (pMsg->power == 3) {
				rc = 1; //收到重连信息
			} else {
				sprintf(payload, "%s", pMsg->allData);
				MQTTMessage message;
				message.qos = QOS1;
				message.retained = false;
				message.payload = (void *) payload;
				message.payloadlen = strlen(payload) + 1;

				if ((rc = MQTTPublish(&client, pubTopic, &message)) != 0) {
					printf("MQTT publish succeed fail %d\n", rc);
				} else {
					printf("MQTT publish succeed  ...\n");
				}
			}
			if (rc != 0) {
				break;
			}
			printf("network: %d\n", rc);
		}
		printf("network.disconnect\n");
		network.disconnect(&network);
		vTaskDelay(1500 / portTICK_RATE_MS);
	}
	printf("mqtt_client_thread going to be deleted\n");
	mHandlerMQTTConntion = NULL;
	vTaskDelete(NULL);
	return;
}

```
而串口打印截图：

```

Task_MqttClient_Connect NetworkConnect init...
 NetworkConnect init succeed...
Use MQTTStartTask succeed...
MQTT subscribe to topic -> 11695776-68c63ab276a0
MqttMessageQueueHandler waitting ..
 [SY] now time: 1 , 20:5:15 , get freeHeap: 42080 , client.isconnected: 1
 [SY] now time: 1 , 20:5:30 , get freeHeap: 42080 , client.isconnected: 1
 [SY] now time: 1 , 20:5:45 , get freeHeap: 42080 , client.isconnected: 1
 [SY] now time: 1 , 20:10:45 , get freeHeap: 42080 , client.isconnected: 1
 [SY] now time: 1 , 20:11:0 , get freeHeap: 42080 , client.isconnected: 1
 [SY] now time: 1 , 20:11:15 , get freeHeap: 42008 , client.isconnected: 0
now send reConnect msg 
 network.disconnect
Task_MqttClient_Connect NetworkConnect init...
 NetworkConnect init succeed...
 [SY] now time: 1 , 20:11:30 , get freeHeap: 41944 , client.isconnected: 0
now send reConnect msg 
 
 [SY] now time: 1 , 20:11:45 , get freeHeap: 41944 , client.isconnected: 0
now send reConnect msg 
 
 [SY] now time: 1 , 20:12:0 , get freeHeap: 41944 , client.isconnected: 0
now send reConnect msg 
```

donghengqaz, 2018-08-28:
你能把工程提供给我们吗？这样我们处理起来效率会高很多。

27-301, 2018-08-28:
@xuhongv Can you raise a PR? so we can review and discuss the code more easily.

xuhongv, 2018-08-28:
[测试包（主工程+lib库文件）.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/2327042/%2Blib.zip)

无法上传完整的工程和SDK包，我只上传了 主工程和lib库文件。

存在着如下问题：

- 第一次连接MQTT服务器成功后，当与路由器断开连接时候，不会自动重连路由器。

- 第一次连接MQTT服务器成功后，当与服务器断开连接时候，发生重连服务器消息，偶尔会一直滞留订阅主题的函数。

xuhongv, 2018-08-30:
请问一下。这个  ``` issue ```有人跟进么？

donghengqaz, 2018-08-30:
这个问题我们这边有专门做 MQTT/SSL 和 底层的人在看。

leon-v, 2018-09-02:
I have been trying to debug this issue for a while. I'm new to pretty much all of this so i doubt ill get to the bottom of it.

But i have tried to disable socket reuse, since i read that could cause this but no success.

I read that there used to be a bug in the FreeRTOS socket send code that set the timeout to max, but that doesn't appear in this.

I also tried to make the RTOS delete the task and re-start it, but it also gets stuck in vTaskDelete() if i do that.

So i believe there is a socket issue somewhere.. I'm just not sure where.

The code i am testing with is here:
https://github.com/leon-v/MQTT-ESP8266-nRF24L01-Solar-Battery-I-O-Network/tree/Socket-Stability-Issues/Router/Firmware/src/components/mqtt_connection

I have also tried LWIP MQTT app, but it didn't have the code to pass the username and password. So i abandon that.

I am not using SSL.

I will try libphy.a version 1055_8

xuhongv, 2018-09-03:
@leon-v Your project is driver on NRF24L ? Nice ! But the rtos MQTT is not so good , is always disconnected on server. I suggest you build on NONOS SDK.

leon-v, 2018-09-03:
I just spent time converting this code from NONOS 😆 
The NRF24L01 driver i just made for this product. But the project is to route NRF24L01 (and other radio modules) packets through to MQTT.

I have been playing with this more, and my thread sometimes locks up when sending to a socket, when making DNS requests as well as MQTT connections.

leon-v, 2018-09-03:
I might try spend time to get on LWIP MQTT client working...

leon-v, 2018-09-03:
Exiting the MQTTRun task when cycle() fails seems to fix my main issue...

`void MQTTRun(void* parm)
{
    Timer timer;
    MQTTClient* c = (MQTTClient*)parm;
    int rc = 0;

    printf("MQTTRun Starting\n");

    TimerInit(&timer);

    while (1) {
        TimerCountdownMS(&timer, 500); /* Don't wait too long if no traffic is incoming */

#if defined(MQTT_TASK)
        MutexLock(&c->mutex);
#endif
        rc = cycle(c, &timer);
#if defined(MQTT_TASK)
        MutexUnlock(&c->mutex);
#endif
        if (rc < 0){
        	printf("MQTTRun Ending\n");
        	break;
        }

    }

    vTaskDelete(NULL);
    return;
}`

I have still had a thread lock up inside `netconn_gethostbyname()`.

But with the above change, i was able to reconnect multiple times successfully.

xuhongv, 2018-09-03:
oh are you have a try ? I test on  2 hours , it have disconnect server and connected failed to server. now wait answer...

xuhongv, 2018-09-11:
@donghengqaz 大半个月过去了，请问此issue有的解决方案了么

xuhongv, 2018-09-15:
@donghengqaz @27-301 能否有个答复呢？

leon-v, 2018-09-16:
I believe the issue is when cyce() sends data into the connected socket at the same time as a publish, which ends up with the MQTT server dropping the client.
To work around this, my latest code will call cycle() and publish in the same loop so they can not run at the same time.

I have also been fighting another issue that only happens at my home when the ESP8266 is connected through a router to a local MQTT server. After some reconnects, the socket no longer works and reconnects and fails infinitely.

leon-v, 2018-09-17:
The code i used wan this :```void MQTTRun(void* parm)
{
    Timer timer;
    MQTTClient* c = (MQTTClient*)parm;
    int rc = 0;
    TimerInit(&timer);
    while (1) {
        TimerCountdownMS(&timer, 500); // Don't wait too long if no traffic is incoming
#if defined(MQTT_TASK)
        MutexLock(&c->mutex);
#endif
        rc = cycle(c, &timer);
#if defined(MQTT_TASK)
        MutexUnlock(&c->mutex);
#endif
        if (rc < 0){
        	break;
        }
    }
    vTaskDelete(NULL);
    return;
}```

Its works better because it closes the thread when there is a connection issue. But its not right.
It should in stead end the thread when MQTT is flagged as not connected.

leon-v, 2018-09-17:
My testing branch is here:
https://github.com/leon-v/MQTT-ESP8266-nRF24L01-Solar-Battery-I-O-Network/blob/Trying-to-get-stable-MQTT-connection/Router/Firmware/src/components/mqtt_connection/mqtt_connection.c

zhanghuichina, 2018-10-12:
networkconnect有时候会一直阻塞，要加个外部watchdog重启这个task

closed 2018-11-06 by xuhongv