espressif/ESP8266_RTOS_SDK #296: printf don't working correct

de1m, 2018-08-31:
## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
__2c72ce9__
- Development Env:      Eclipse and direct in linux console
- Operating System:    Ubunutu OS and Windows 10 Notebook
- Power Supply:         PC USB

## Problem Description
I've created a simple example to print out in UART a message 
//Detailed problem description goes here.

### Expected Behavior
Normal UART output (I use the cutecom in Linux and hterm in Windows)

### Actual Behavior
Output show like
```
<0x0c>bp<0x82>lb<0xec><0x12>r<0x82><0x92><0x92>n<0x92><0x02><0x12>n<0x1e><break>
```
Full output: [gist](https://gist.github.com/de1m/4a813b6a292ae55b7d3ef7865955b6e5)

[sdkconfig.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/2340058/sdkconfig.txt)
### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "nvs.h"

void app_main()
{
    // Restart module
    for (int i = 10; i >= 0; i--) {
        printf("Restarting in seconds...\n");
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```
I use the modul NodeMCU ESP8266 ESP-12E V1.0

xiebadegg, 2018-08-31:
What is your baud rate?

de1m, 2018-08-31:
@xiebadegg, this is configured in menuconfig - 115200

xiebadegg, 2018-09-01:
## Did your 8266 print the paragraph on the boot?
```
<0x0c>bp<0x82>lb<0xec><0x12>r<0x82><0x92><0x92>n<0x92><0x02><0x12>n<0x1e><break>
```
## And no printing.
```
Restarting now.\\n
```
##  8266 print boot messages on boot time, and the boot message baud rate is 74880
##  Modified baud rate is 74880, let me see the print results.

xiebadegg, 2018-09-01:
Sorry, I may have misunderstood you.

de1m, 2018-09-01:
The baud rate change from 115200 to 74880 has solved my problem

closed 2018-09-01 by de1m