espressif/ESP8266_RTOS_SDK #297: 关于GPIO问题，同样的配置4个输出的GPIO，只有一个正常。

tianhuanz, 2018-09-02:
让我觉得奇怪的是，修改GPIO配置，OTA升级后（我确定升级成功），竟然需要给设备重新上电才生效。
    OTA升级的复位和上电复位有多大却别？我可以明显感觉到OTA复位的速度比上电复位速度要快很多。

代码如下：
    gpio_set_level(PLUG_SWITCH_GPIO, switchState);
    gpio_set_level(PLUG_ALARM_LED_GPIO, 1);
    gpio_set_level(PLUG_STATUS_LED_GPIO, !switchState);
    gpio_set_level(PLUG_BUZZER_GPIO, 0);
    gpio_set_direction(PLUG_SWITCH_GPIO, GPIO_MODE_OUTPUT);
    gpio_set_direction(PLUG_ALARM_LED_GPIO, GPIO_MODE_OUTPUT);
    gpio_set_direction(PLUG_STATUS_LED_GPIO, GPIO_MODE_OUTPUT);
    gpio_set_direction(PLUG_BUZZER_GPIO, GPIO_MODE_OUTPUT);
    gpio_set_direction(PLUG_BUTTON_GPIO, GPIO_MODE_INPUT);

xiongyumail, 2018-09-03:
Hi @tianhuanz 
能否提供一些更详细的信息。gpio example可以跑通嘛？

FayeY, 2018-09-20:
你好，由于长时间无回复，此问题暂时关闭，如果有任何新进展，欢迎随时 reopen，烦请提供更详细的 debug 信息以供查证。

closed 2018-09-20 by FayeY