espressif/ESP8266_RTOS_SDK #299: v3 的文档什么时候出来呀

shangzhe, 2018-09-03:
好多问题

xiongyumail, 2018-09-03:
Hi @shangzhe 
请问具体是什么问题？

环境搭建可以参考[README](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/README.md)

shangzhe, 2018-09-04:
ESP8266_RTOS_SDK\components\esp8266\include\rom\ets_sys.h

/**
  * @brief     Register the print output function.
  *
  * @attention os_install_putc1((void *)uart1_write_char) in uart_init will set
  *            printf to print from UART 1, otherwise, printf will start from
  *            UART 0 by default.
  *
  * @param     void(*p)(char c) - pointer of print function
  *
  * @return    null
  */
void os_install_putc1(void (*p)(char c));

但是v3 的sdk根本就没有 uart1_write_char函数 我把函数从v2那边复制过来，也不行 。
找半天才找到在这里改
ESP8266_RTOS_SDK\components\esp8266\source\ets_printf.c
#ifndef CONFIG_ETS_PUTC_UART
#define CONFIG_ETS_PUTC_UART 1
#endif

FayeY, 2018-10-23:
非常抱歉，目前的文档还不够完善，开发文档可见 [ESP8266_RTOS_SDK/docs](https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/docs/en)，我们还在继续完善中。
请知悉，谢谢！

closed 2018-10-23 by FayeY