espressif/ESP8266_RTOS_SDK #305: ESP8266 logging and printing on UART1 (realease/v3.0)

Bennji94, 2018-09-05:
Hello, is it possible to redirect logging output of ESP_LOGI or printf to UART1 instead of UART0?

In former releases there was the option with SetPrintPort and for the ESP32 you can set the UART with make menuconfig. But both are not available for the ESP8266 with the RTOS_SDK v3.0.

Best regards

invention88, 2018-09-06:
can refer to:
(2) The CONFIG_ETS_PUTC_UART serial port number is defined in Ets_printf.c. Can be switched dynamically.
(3) UART1 (GPIO2) cannot be connected to the serial line at power-on. Otherwise a system exception will occur.

27-301, 2018-09-07:
@Bennji94 Yes, we will add this into menuconfig soon.

27-301, 2018-09-14:
@Bennji94 @invention88 It's now supported in SDK, you can use menuconfig to choose the serial port which you want to output the log.

Component config > ESP8266-specific > UART peripheral to use for console output (0-1)

closed 2018-09-14 by 27-301