espressif/ESP8266_RTOS_SDK #308: The newest ESP8266_RTOS_SDK build error

xiaolongba, 2018-09-10:
Hi,all
     I build the project_template project,and it error occurs.This is my reproduce step as follows：
- 1.Download the ESP8266_RTOS_SDK by "git clone https://github.com/espressif/ESP8266_RTOS_SDK.git"

- 2.Download the toolchain of esp8266
[image]

- 3.Setup the variable of PATH and IDF_PATH in the "D:\Project\esp8266\msys32\etc\profile.d\esp32_toolchain.sh",the context as follows:
```
export PATH="$PATH:/opt/xtensa-lx106-elf/bin"
export IDF_PATH="$HOME/esp/ESP8266_RTOS_SDK"
```

- 4.Identify the PATH and IDF_PATH
[image]

- 5.After setting the above steps and build the demo project,error occur and the log as follows:
```
Administrator@Helon MINGW32 ~/esp/ESP8266_RTOS_SDK/examples/get-started/project_template
$ make all -j8
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
CC build/app_update/esp_ota_ops.o
CC build/bootloader_support/src/bootloader_random.o
CC build/cjson/cJSON/cJSON.o
CC build/bootloader_support/src/flash_encrypt.o
CC build/components/folder1/file1.o
CC build/esp8266/driver/gpio.o
CC build/espos/platform/freertos/espos_timer.o
AR build/app_update/libapp_update.a
CC build/bootloader_support/src/bootloader_sha.o
CC build/cjson/cJSON/test.o
CC build/components/folder2/file2.o
CC build/bootloader_support/src/esp_image_format.o
CC build/esp8266/driver/uart.o
D:\Project\esp8266\msys32\opt\xtensa-lx106-elf\bin\xtensa-lx106-elf-ar.exe: esp_ota_ops.o: No such file or directory
CC build/espos/platform/freertos/espos_queue.o
make[1]: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:215：libapp_update.a] 错误 1
make: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/make/project.mk:467：component-app_update-build] 错误 2
make: *** 正在等待未完成的任务....
CC build/bootloader_support/src/flash_partitions.o
CC build/espos/platform/freertos/espos_task.o
CC build/cjson/cJSON/cJSON_Utils.o
AR build/components/libcomponents.a
CC build/esp8266/source/startup.o
CC build/esp8266/source/esp_socket.o
CC build/espos/platform/freertos/espos_mutex.o
D:\Project\esp8266\msys32\opt\xtensa-lx106-elf\bin\xtensa-lx106-elf-ar.exe: folder1/file1.o: No such file or directory
CC build/bootloader_support/src/secure_boot_signatures.o
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
make[1]: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:215：libcomponents.a] 错误 1
AR build/cjson/libcjson.a
CC build/esp8266/source/event_loop.o
make: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/make/project.mk:467：component-components-build] 错误 2
CC build/bootloader/main/bootloader_start.o
CC build/esp8266/source/chip_boot.o
CC build/espos/platform/freertos/espos_scheduler.o
CC build/bootloader_support/src/bootloader_clock.o
D:\Project\esp8266\msys32\opt\xtensa-lx106-elf\bin\xtensa-lx106-elf-ar.exe: cJSON/cJSON.o: No such file or directory
CC build/esp8266/source/esp_wifi_os_adapter.o
make[1]: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:215：libcjson.a] 错误 1
make: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/make/project.mk:467：component-cjson-build] 错误 2
CC build/bootloader_support/src/secure_boot.o
CC build/espos/platform/freertos/espos_semaphore.o
AR build/bootloader/main/libmain.a
CC build/esp8266/source/ets_printf.o
CC build/esp8266/source/esp_wifi.o
CC build/espos/platform/freertos/espos_time.o
CC build/bootloader_support/src/bootloader_common.o
D:\Project\esp8266\msys32\opt\xtensa-lx106-elf\bin\xtensa-lx106-elf-ar.exe: bootloader_start.o: No such file or directory
CC build/esp8266/source/phy_init.o
make[2]: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:215：libmain.a] 错误 1
CC build/bootloader_support/src/bootloader_utility.o
make[1]: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/make/project.mk:467：component-main-build] 错误 2
make: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/components/bootloader/Makefile.projbuild:40：/home/Administrator/esp/ESP8266_RTOS_SDK/examples/get-started/project_template/build/bootloader/bootloader.bin] 错误 2
CC build/bootloader_support/src/bootloader_init.o
CC build/esp8266/source/event_default_handlers.o
AR build/espos/libespos.a
CC build/esp8266/source/system_api.o
CC build/bootloader_support/src/efuse.o
CC build/bootloader_support/src/flash_qio_mode.o
D:\Project\esp8266\msys32\opt\xtensa-lx106-elf\bin\xtensa-lx106-elf-ar.exe: platform/freertos/espos_timer.o: No such file or directory
make[1]: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:215：libespos.a] 错误 1
make: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/make/project.mk:467：component-espos-build] 错误 2
CC build/bootloader_support/src/bootloader_flash.o
AR build/esp8266/libesp8266.a
D:\Project\esp8266\msys32\opt\xtensa-lx106-elf\bin\xtensa-lx106-elf-ar.exe: driver/gpio.o: No such file or directory
AR build/bootloader_support/libbootloader_support.a
make[1]: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:215：libesp8266.a] 错误 1
make: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/make/project.mk:467：component-esp8266-build] 错误 2
D:\Project\esp8266\msys32\opt\xtensa-lx106-elf\bin\xtensa-lx106-elf-ar.exe: src/bootloader_random.o: No such file or directory
make[1]: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:215：libbootloader_support.a] 错误 1
make: *** [/home/Administrator/esp/ESP8266_RTOS_SDK/make/project.mk:467：component-bootloader_support-build] 错误 2
```

 Can somebody help me?Or give me some suggestions.Thanks.

zhanghuichina, 2018-09-20:
same question

zhanghuichina, 2018-09-20:
xtensa-lx106-elf-gcc.exe this file size is 0, wrong here, where i can get other  version ffor windows

27-301, 2018-09-21:
hi, how do you extra xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz? I find in windows, when use tool such as winrar, the size if 0. Please try to use command line to extra this file, `tar zxvf xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz -C /opt`.

zhanghuichina, 2018-09-21:
https://dl.espressif.com/dl/xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz  this is ok

xiaolongba, 2018-09-22:
@27-301 
Thanks,the issue sloved,u are a good man.the xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz must be extraed in linux os by tar command line.

closed 2018-09-25 by FayeY