espressif/ESP8266_RTOS_SDK #309: ipv6 in latest SDK

nuclearcat, 2018-09-11:
I gitcloned latest SDK and it works fine for ipv4, but now i am trying to setup ipv6 over SLAAC
I have tested ipv6 in my network by several "usual" devices (windows, linux), so it is working, it is quite trivial juniper srx config.

ESP8266 code that seems to start working, and getting some results:
on SYSTEM_EVENT_STA_GOT_IP
i do:
interface->ip6_autoconfig_enabled=1;
netif_create_ip6_linklocal_address(interface, 1);

It seems even reach stage of getting global ipv6
netif: IPv6 address 2 of interface en set to XXXX:XXX:0:4000:5ECF:7FFF:FE85:E0B4/0x30

but unfortunately router is not seeing it as assigned in ipv6 neighbours.

I am getting 3 ips, link local, ULA and global, relevant log entries:
netif: IPv6 address 0 of interface en set to FE80::5ECF:7FFF:FE85:E0B4/0x30
netif: IPv6 address 1 of interface en set to FDDB:D7BD:695F:0:5ECF:7FFF:FE85:E0B4/0x30
netif: IPv6 address 2 of interface en set to XXXX:XXX:0:4000:5ECF:7FFF:FE85:E0B4/0x30

It is pingable from router:
> ping FE80::5ECF:7FFF:FE85:E0B4               
PING6(56=40+8+8 bytes) fe80::8ae0:f3ff:fe28:5a90 --> fe80::5ecf:7fff:fe85:e0b4
16 bytes from fe80::5ecf:7fff:fe85:e0b4, icmp_seq=0 hlim=255 time=219.650 ms
16 bytes from fe80::5ecf:7fff:fe85:e0b4, icmp_seq=1 hlim=255 time=385.116 ms

But not pingable from computers in same network. on ESP8266 i enabled some debug, and can see at this moment:
ip6_input: packet not for us.
ip6_input: packet not for us.
ip6_input: packet not for us.

Here is what computer sending:
22:08:38.826869 IP6 (hlim 255, next-header ICMPv6 (58) payload length: 32) fe80::508c:dd12:80ef:7003 > ff02::1:ff85:e0b4: [icmp6 sum ok] ICMP6, neighbor solicitation, length 32, who has fe80::5ecf:7fff:fe85:e0b4
	  source link-address option (1), length 8 (1): 4c:cc:6a:f4:XX:XX
	    0x0000:  4ccc 6af4 342e

It is getting global and ULA address, but seems configuration is not completing, as it is not seen in router.
> show ipv6 neighbors | match 5c:cf              
fe80::5ecf:7fff:fe85:e0b4    5c:cf:7f:85:e0:b4  stale       739 no  no      vlan.0      
>

Please let me know if you need more information.
If you can make ipv6 working with ESP8266 - it will be amazing, and i think it will make Espressif absolute leader in IoT MCU market!

nuclearcat, 2018-09-12:
Ok, i found way to make ipv6 fully working, main reason is lack of examples :)
Reason it was not answering and not appearing, because it is not listening packets with destination  solicited-Node multicast address by default, and it is not listening to it, because it needs specific flag set on interface - NETIF_FLAG_MLD6.

`    case SYSTEM_EVENT_STA_GOT_IP:
        xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
	struct netif *interface = netif_list;
	interface->ip6_autoconfig_enabled=1;
	interface->flags      |= NETIF_FLAG_MLD6;
	netif_create_ip6_linklocal_address(interface, 0);
        break;
`

27-301, 2018-09-14:
@Junhao-Espressif Please check whether it can be improved in SDK side, and prepare some ipv6 examples. Thanks.

closed 2018-09-25 by wujiangang