espressif/ESP8266_RTOS_SDK #310: How do i mofiy the partition table of the cutomer mode

invention88, 2018-09-12:
Such as the title.
When using the customer partition mode, what are the rule restrictions when configuring the partition table? Includes single app and tow ota modes.

27-301, 2018-09-14:
hi, you can refer to https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/docs/en/api-guides/partition-tables.rst

invention88, 2018-09-15:
I set custom partition table and modif ota_0 size to 0xE0000, modify ota_1 offset to 0x100000 in partition.cvs file, like this:
ota_0 | 0x10000 | 0xE0000
ota_1 | 0x100000 | 0xE0000

After compiling successfully, burn to flash to run the ota example. The app of oat_1 cannot be run. The system is stuck in the Cache_Read_Enable of the bootloader_utility_load_image function.

invention88, 2018-09-15:
I think it is free to configure the starting address and size of each segment of data and app. (by sector size)

invention88, 2018-09-15:
I think it is possible to arbitrarily configure the starting address and size of each segment of data and app. (by sector size)

ustccw, 2018-09-17:
@27-301 Hi, start addr of `ota_0` and `ota_1` can not modify because of flash cache. the size of `ota_0` and `ota_1` can be modified. 
once user reduce the ota size, customer partition can be set to the free sectors.

ustccw, 2018-09-17:
@invention88  Hi, start addr of ota_0 and ota_1 can not modify because of flash cache. the size of ota_0 and ota_1 can be modified.
once user reduce the ota size, customer partition can be set to the free sectors.

invention88, 2018-09-17:
@ustccw got it, thanks

closed 2018-09-18 by FayeY