espressif/ESP8266_RTOS_SDK #314: Some Kconfig options are mandatory

THKDev, 2018-09-16:
## Environment

- Development Kit: ESP-12E
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //1a6460b19a0df1ee0aff087a1f37edd3304851d3
- Development Env:     Make
- Operating System:     Fedora
- Power Supply:         external 3.3V

## Problem Description
- Compile options -> Assertion level -> Disabled
Programm won't start anymore
- Component Config -> FreeRTOS -> Tick Rate (Hz) -> set to 400
Free RTOS timer are out of sync by many seconds (~15 seconds)
- Component Config -> **disable** Newlib  (unselect)
  many compile and link errors
- Component Config -> LWIP -> SOCKET -> **disable** LWIP socket UDP sync send (unselect)
  compile error: missing function
- Component Config -> **disable** Initialize Task Watchdog Timer on startup (unselect)
  compile error if log level is set to DEBUG

donghengqaz, 2018-09-18:
Thanks a lot and some problems are solved.

```
Component Config -> FreeRTOS -> Tick Rate (Hz) -> set to 400
Free RTOS timer are out of sync by many seconds (~15 seconds)
```

How can I understand this ?

THKDev, 2018-09-18:
You are right, i forget some information. This is the code i use

```
static void observerTask(void *pvParameter) 
{
    while (true) {
        vTaskDelay(60 * 1000 / portTICK_RATE_MS);
        const uint32_t heap = heap_caps_get_free_size(caps);
        time_t     rawtime;
        struct tm  lTime;

        time(&rawtime);
        localtime_r(&rawtime, &lTime);
        strftime(buf, bufSize, "%c", &lTime);
        LOGI("Date: %s, minheap: %d, heap: %d", asclocaltime(asctimeBuffer, sizeof(asctimeBuffer)), heap_caps_get_minimum_free_size(caps), heap);
    }
}
```
And this is the monitor output if tick rate is set to 400.
```
2018-09-18 17:09:06.803:  I (120243) observerTask: Date: Thu Jan  1 00:02:28 1970, minheap: 37764, heap: 41184
2018-09-18 17:10:21.813:  I (180246) observerTask: Date: Thu Jan  1 00:03:44 1970, minheap: 37764, heap: 41184
2018-09-18 17:11:36.820:  I (240254) observerTask: Date: Thu Jan  1 00:04:59 1970, minheap: 37764, heap: 41184
2018-09-18 17:12:51.831:  I (300258) observerTask: Date: Thu Jan  1 00:06:14 1970, minheap: 37764, heap: 41184

```

donghengqaz, 2018-10-09:
Oh， I has worked out the reason, when you set tick rate to be 400, then:

```
-> configTICK_RATE_HZ = 400
-> portTICK_PERIOD_MS = 1000 / 400 = 2
-> vTaskDelay(60 * 1000 / portTICK_RATE_MS) = vTaskDelay(60 * 1000 / 2) = vTaskDelay(30000)
```

But real tick cycle is 2.5ms, sleep time = 2.5 * 0.001 * 30000 = 75 s, 75 - 60 = 15s.

That's why it always sleep extra 15s.

We should add note to the Kconfig file to let user know the value must be The common divisor of 1000.

Thanks a lot.

closed 2018-10-11 by wujiangang