espressif/ESP8266_RTOS_SDK #317: error log: bootloader_start.o ->No such file or directory

imliubo, 2018-09-17:
## Environment

- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 1a6460b
- Development Env:     Make
- Operating System:     Windows

## Problem Description
```bash
make all
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
G:\espressif\msys32\opt\xtensa-lx106-elf\bin\xtensa-lx106-elf-ar.exe: bootloader_start.o: No such file or directory
make[2]: *** [/home/LIUBO/ESP8266_RTOS_SDK/make/component_wrapper.mk:215：libmain.a] 错误 1
make[1]: *** [/home/LIUBO/ESP8266_RTOS_SDK/make/project.mk:467：component-main-build] 错误 2
make: *** [/home/LIUBO/ESP8266_RTOS_SDK/components/bootloader/Makefile.projbuild:40：/home/LIUBO/projects/project_template/build/bootloader/bootloader.bin] 错误 2
```

### Expected Behavior
make all the components successfully
### Actual Behavior
G:\espressif\msys32\opt\xtensa-lx106-elf\bin\xtensa-lx106-elf-ar.exe: bootloader_start.o: No such file or directory

I check the directory and it without bootloader_start.o ，the **make all** command should Compile and generate the bootloader_start.o file.
### Steps to repropduce

1. Download the Windows all-in-one toolchain & MSYS2 zip file from dl.espressif.com:
[https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20180110.zip](https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20180110.zip)
2. Download the toolchain of esp8266:
[https://dl.espressif.com/dl/xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz](https://dl.espressif.com/dl/xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz)
3. Download ESP8266_RTOS_SDK:
```bash
git clone https://github.com/espressif/ESP8266_RTOS_SDK.git
```
4. Setup Path to ESP8266_RTOS_SDK as this file:
```sh
export PATH="$PATH:/opt/xtensa-lx106-elf/bin"
export IDF_PATH="G:/espressif/msys32/home/LIUBO/ESP8266_RTOS_SDK"
```
5. Confirm PATH and IDF_PATH:
```bash
$ printenv PATH
/mingw32/bin/:/mingw32/bin:/usr/local/bin:/usr/bin:/bin:/c/Windows/System32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0/:**/opt/xtensa-lx106-elf/bin**:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl

$ printenv IDF_PATH
**G:/espressif/msys32/home/LIUBO/ESP8266_RTOS_SDK**
```
6. After all setting,then I build example for test:
```bash
$ make all
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
G:\espressif\msys32\opt\xtensa-lx106-elf\bin\xtensa-lx106-elf-ar.exe: bootloader_start.o: No such file or directory
make[2]: *** [/home/LIUBO/ESP8266_RTOS_SDK/make/component_wrapper.mk:215：libmain.a] 错误 1
make[1]: *** [/home/LIUBO/ESP8266_RTOS_SDK/make/project.mk:467：component-main-build] 错误 2
make: *** [/home/LIUBO/ESP8266_RTOS_SDK/components/bootloader/Makefile.projbuild:40：/home/LIUBO/projects/project_template/build/bootloader/bootloader.bin] 错误 2
```

zhanghuichina, 2018-09-20:
I have the same question, how to fix it?

zhanghuichina, 2018-09-20:
xtensa-lx106-elf-gcc.exe   this file size is 0, wrong here

imliubo, 2018-09-21:
> xtensa-lx106-elf-gcc.exe this file size is 0, wrong here

> I have the same question, how to fix it?

still waiting the engineer ~

27-301, 2018-09-21:
hi, how do you extra xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz? I find in windows, when use tool such as winrar, the size if 0. Please try to use command line to extra this file, tar zxvf xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz -C /opt.

imliubo, 2018-09-21:
> hi, how do you extra xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz? I find in windows, when use tool such as winrar, the size if 0. Please try to use command line to extra this file, tar zxvf xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz -C /opt.

Thanks to reply.Yes,I use 360zip to extra the file,I  will try extra the file with command line,hope it can work~

imliubo, 2018-09-21:
> hi, how do you extra xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz? I find in windows, when use tool such as winrar, the size if 0. Please try to use command line to extra this file, tar zxvf xtensa-lx106-elf-win32-1.22.0-88-gde0bdc1-4.8.5.tar.gz -C /opt.

Yes,It **works**.You are right,I compare the extra file with 360zip and the command line ,Two files are not the same size.Thank you for solving my problem.Have nice day~

closed 2018-09-21 by imliubo