espressif/ESP8266_RTOS_SDK #318: Race condition at tcp_output

THKDev, 2018-09-18:
## Environment

- Development Kit: ESP-12E
- IDF version (``git rev-parse --short HEAD`` to get the commit id.):  e409678ccc6399a1b1ca4656267f507f6a116fc5
- Development Env:      Make
- Operating System:     Fedora
- Power Supply:         external 3.3V

## Problem Description
Race condition between calling tcp_output(pcb) and tcp_rexmit_rto(pcb) called by tcp_slowtmr(). 
This issue seems to be solved with version 2.1.x of LWIP.

### Expected Behavior
No race condition.

### Actual Behavior
TCP stack eventual crash by calling tcp_output to flush tcp packages.

### Steps to repropduce
Use [MQTT implementation](https://github.com/espressif/ESP8266_RTOS_SDK/blob/e409678ccc6399a1b1ca4656267f507f6a116fc5/components/lwip/lwip/src/apps/mqtt/mqtt.c#L237) provided by LWIP.

## Debug Logs

```
2018-09-18 18:09:57.409:  tcp_write(pcb=0x401072a0, data=0x3ffedeb5, len=104, apiflags=1)
2018-09-18 18:09:57.418:  tcp_write: queueing 62707:62811
2018-09-18 18:09:57.422:  tcp_output_segment: 62707:62811
2018-09-18 18:09:57.426:  tcp_output: nothing to send (0)
2018-09-18 18:09:57.617:  tcp_output: sending ACK for 700458022
2018-09-18 18:09:57.625:  tcp_output: nothing to send (0)
2018-09-18 18:09:57.638:  tcp_write(pcb=0x401072a0, data=0x3ffedf1d, len=19, apiflags=3)
2018-09-18 18:09:57.644:  tcp_write: queueing 62811:62830
2018-09-18 18:09:57.646:  tcp_write(pcb=0x401072a0, data=0x3ffede30, len=85, apiflags=1)
2018-09-18 18:09:57.657:  tcp_output_segment: 62811:62915
2018-09-18 18:09:57.659:  tcp_output: nothing to send (0)
2018-09-18 18:09:57.663:  tcp_write(pcb=0x401072a0, data=0x3ffede85, len=104, apiflags=1)
2018-09-18 18:09:57.674:  tcp_write: queueing 62915:63019
2018-09-18 18:09:57.676:  tcp_output_segment: 62915:63019
2018-09-18 18:09:57.680:  tcp_output: nothing to send (0)
2018-09-18 18:09:57.867:  tcp_output: sending ACK for 700458101
2018-09-18 18:09:57.877:  tcp_output: sending ACK for 700458172
2018-09-18 18:09:57.888:  tcp_write(pcb=0x401072a0, data=0x3ffedeed, len=67, apiflags=3)
2018-09-18 18:09:57.894:  tcp_write: queueing 63019:63086
2018-09-18 18:09:57.896:  tcp_write(pcb=0x401072a0, data=0x3ffede30, len=37, apiflags=1)
2018-09-18 18:09:57.907:  tcp_output_segment: 63019:63123
2018-09-18 18:09:57.909:  tcp_output_segment: 63019:63123
2018-09-18 18:09:57.913:  
2018-09-18 18:09:57.918:  
2018-09-18 18:09:57.918:  Task stack [MqttDispatcher] stack from [3fff5b34] to [3fff6334], total [2052] size
2018-09-18 18:09:57.926:  
2018-09-18 18:09:57.926:            00000000 00000001 00000002 00000003 00000004 00000005 00000006 00000007 00000008 00000009 0000000a 0000000b 0000000c 0000000d 0000000e 0000000f 
2018-09-18 18:09:57.948:  
2018-09-18 18:09:57.948:  3fff5b34  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
2018-09-18 18:09:57.969:  3fff5b74  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
2018-09-18 18:09:57.990:  3fff5bb4  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
2018-09-18 18:09:58.012:  3fff5bf4  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
2018-09-18 18:09:58.033:  3fff5c34  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
2018-09-18 18:09:58.054:  3fff5c74  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
2018-09-18 18:09:58.072:  3fff5cb4  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
2018-09-18 18:09:58.093:  3fff5cf4  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
2018-09-18 18:09:58.114:  3fff5d34  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
2018-09-18 18:09:58.136:  3fff5d74  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
2018-09-18 18:09:58.157:  3fff5db4  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
2018-09-18 18:09:58.178:  3fff5df4  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 
2018-09-18 18:09:58.200:  3fff5e34  a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5 40105904 
2018-09-18 18:09:58.221:  3fff5e74  401002aa 00000030 4023cbe2 3fff5ec0 00000020 00000000 40106820 40106820 3ffe96e8 00000000 00000003 40105904 401002aa 00000030 4023cbe2 3fff5ef0 
2018-09-18 18:09:58.238:  3fff5eb4  00000020 00000000 40106820 40106820 3ffe95f0 00000000 00000003 00000000 00000000 00000000 3fff04e8 00000000 00000000 00000000 0000001d 00000000 
2018-09-18 18:09:58.260:  3fff5ef4  00000000 3fff04e8 4023cbe2 3fff5f40 00000020 00000000 40106968 0000000a 3fff5f50 00000000 00000000 00000000 00000000 00000000 4023e535 3fff5f70 
2018-09-18 18:09:58.281:  3fff5f34  00000020 00000000 40105510 40105904 401002aa 00000030 4023cbe2 3fff5f90 00000020 00000000 40106820 40106820 3ffe96f0 00000000 00000003 00000000 
2018-09-18 18:09:58.302:  3fff5f74  00000000 00000000 3fff04e8 00000000 00000000 00000000 0000001d 00000000 00000000 3fff04e8 4023cbe2 3ffe87f0 3ffeecf2 00000000 4023e530 0000000a 
[... 8 lines cut ...]
2018-09-18 18:09:58.490:  3fff61b4  00000000 00000000 401072a4 401071b0 401072a0 40107450 40246a0c 401071b0 00000000 00000000 40247757 00000000 401072a0 40107450 402427e0 401071b0 
2018-09-18 18:09:58.512:  3fff61f4  00000000 00000018 00000050 3ffeed38 3ffe9648 00000000 00000003 401072a0 000005b4 00000001 3ffeed38 00000018 401072a0 40107450 402431b9 00000000 
2018-09-18 18:09:58.533:  3fff6234  00007540 401071b0 00000025 00000000 00000000 00000000 401072a0 00000025 00000b25 3ffede2c 4024a73c 00000001 401002aa 00000030 4024a329 3fff62b0 
2018-09-18 18:09:58.554:  3fff6274  00000020 00000000 402286a8 00000066 3ffede2c 3ffedd40 4024b0a2 00000001 00000045 0000001d 3ffedd5c 40107378 00000000 00000221 40107434 00000000 
2018-09-18 18:09:58.572:  3fff62b4  00000000 00000000 3ffedd40 402286a8 40107378 3ffedd2c 40249fef 40249d00 3ffedd2c 00000002 401002f3 3ffedf44 3ffedf3c 00000001 00000002 3ffedf44 
2018-09-18 18:09:58.593:  3fff62f4  3ffedf3c 00000000 4024a088 3ffedd2c 401073d0 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 
2018-09-18 18:09:58.614:  3fff6334  bfff5b2c 
2018-09-18 18:09:58.618:  
2018-09-18 18:09:58.618:  
2018-09-18 18:09:58.619:        PC: 0x4024329d       PS: 0x00000030       A0: 0x40243299       A1: 0x3fff6230 
0x4024329d: tcp_output at /home/user/projects/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/tcp_out.c:1154

0x40243299: tcp_output at /home/user/projects/esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/tcp_out.c:1154

2018-09-18 18:09:58.639:        A2: 0x00000000       A3: 0x0000f600       A4: 0x0000f600       A5: 0x00000000 
2018-09-18 18:09:58.640:        A6: 0x3ffe96f0       A7: 0x00000000       A8: 0x00000003       A9: 0x00000000 
2018-09-18 18:09:58.652:       A10: 0x00000000      A11: 0x00000000      A12: 0x40107450      A13: 0x401072a0 
2018-09-18 18:09:58.670:       A14: 0x0000f62b      A15: 0x00000068      SAR: 0x0000001d EXCCAUSE: 0x0000001c 
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Junhao-Espressif, 2018-09-20:
Hi @THKDev 

Could you provide demo that you wrote with  lwip mqtt ?

Thanks!

Junhao-Espressif, 2018-09-26:
Hi @THKDev 

Could you tell me the mqtt server name?

How to config ```Hostname of the MQTT server```

Thanks!

THKDev, 2018-09-27:
@Junhao-Espressif 
you can set the name of the MQTT server by `make menuconfig` and then `Project configuration  ---> Hostname of the MQTT server`

Junhao-Espressif, 2018-09-28:
Hi @THKDev 

Sorry, I didn't describe it clearly.

I want to know the mqtt server name you set.

Thanks!

THKDev, 2018-09-29:
I've been using the hostname of my mosquitto mqtt server.

Junhao-Espressif, 2018-10-10:
Hi @THKDev 

line 1038 of tcp_output.c: ```useg = pcb->unacked``` 
    useg is null here.

line 1146 of tcp_output.c: ```if (pcb->unacked == NULL) {``` 
    pcb->unacked is not null here because of TCP retransmission.  
    But useq is null and crashed.

This problem arises if you call tcp_output directly in application in multithreaded operating system. Please use lwip_send or socket instead of tcp_output.  
You can add code like this: 
```      } else {
        /* useg should point to last segment on unacked queue */
        useg = pcb->unacked;
        if (useg != NULL) {
          for (; useg->next != NULL; useg = useg->next);
        }
        /* In the case of fast retransmit, the packet should not go to the tail
         * of the unacked queue, but rather somewhere before it. We need to check for
         * this case. -STJ Jul 27, 2004 */````

Could you use our MQTT code which based on socket？

THKDev, 2018-10-10:
@Junhao-Espressif 
thanks for your reply. But i do not understand your fix. So i decided to take another solution. I enabled [LWIP_TCPIP_CORE_LOCKING](https://github.com/espressif/ESP8266_RTOS_SDK/blob/93e9b1c3d34de32da798151996a396fe0a104e54/components/lwip/port/esp8266/include/lwipopts.h#L163) and [lock](https://github.com/espressif/ESP8266_RTOS_SDK/blob/93e9b1c3d34de32da798151996a396fe0a104e54/components/lwip/lwip/src/include/lwip/tcpip.h#L56) the TCPIP core before any mqtt message and [unlock](https://github.com/espressif/ESP8266_RTOS_SDK/blob/93e9b1c3d34de32da798151996a396fe0a104e54/components/lwip/lwip/src/include/lwip/tcpip.h#L58) it afterwards. This solves the race condition at all.

You mean the Paho mqtt implementation? I tried it. But connection and error handling in case of network or server problems are insufficient.
The LwIP implementation is very stable and callbacks working perfect in any case of error.

Junhao-Espressif, 2018-10-11:
Hi @THKDev 

I think the application layer and the lwip core are best separated from each other and try not to call the LwIP underlying interface, such as ```tcp_output```.
If there is no other problem, please close this issue. 
Thank you!

THKDev, 2018-10-11:
That's right. But [mqtt.c](https://github.com/espressif/ESP8266_RTOS_SDK/blob/c4f2a58351b362640f7eb9bf8df147e83970b50c/components/lwip/lwip/src/apps/mqtt/mqtt.c#L1) is part of the lwIP core. Some maybe [LWIP_TCPIP_CORE_LOCKING](https://github.com/espressif/ESP8266_RTOS_SDK/blob/93e9b1c3d34de32da798151996a396fe0a104e54/components/lwip/port/esp8266/include/lwipopts.h#L163) should be enabled.
See also [a note](http://lwip.100.n7.nabble.com/LWIP-TCPIP-CORE-LOCKING-option-tp27834p27835.html) by Simon Goldschmidt (project admin of lwIP).

> The core code of lwIP is not protected against concurrent access. As such, "API" code has
> to take care of multithreading. The old method was message-passing (i.e. send -> message
> to tcpip_thread -> tcpip_thread executes the send and wakes up the API thread when done).
> Now when enabling CORE_LOCKING, we use a global mutex instead of message passing.
> 
> Both versions work but with current operating systems, the mutex version should be faster
> since in the case when noone else is working with the core, grabbing a mutex is so much
> faster than 2 thread context switches. 
>

closed 2018-10-11 by THKDev