espressif/ESP8266_RTOS_SDK #321: FreeRTOS update to 10.0.1. MIT License

gregorHerburger, 2018-09-21:
Dear Sirs,

for our project we want to use your ESP8266_RTOS_SDK. 
But we cannot use GPL License in our Project. In the provided SDK the FreeRTOS Source Files are for a better part in the new version 10.0.1 which is licensed under the MIT License. But unfortanately the following files: 
components/freertos/port/esp8266/heap_5.c
components/freertos/port/esp8266/port.c
components/freertos/port/esp8266/include/freertos/FreeRTOSConfig.h
components/freertos/port/esp8266/include/freertos/portmacro.h

are from an older version, which is licensed under the GPL license. 

As it is crucial for our project to not use any GPL license files, could you update the above mentioned files to the actual version (10.0.1) which is licensed under MIT?

Best regards.

27-301, 2018-09-25:
Yes, it can be updated to MIT license. @donghengqaz

donghengqaz, 2018-10-09:
We will modify these license at v3.1 version.

gregorHerburger, 2018-10-09:
Thank You. Good News. 

When is this version v3.1 expected to be released?

donghengqaz, 2018-10-10:
We planed to release v3.1 at the end of October。

donghengqaz, 2018-10-11:
The license at master branch has been changed.

closed 2018-10-11 by wujiangang