espressif/ESP8266_RTOS_SDK #323: IDF_PATH -- same for ESP8266 and ESP32

jhinklejr, 2018-09-27:
If developing for both the ESP32 and ESP8266 RTOS using the new IDF process how do you recommend have the same env variable IDF_PATH but with different associated path strings?

Should the IDF_PATH variable be changes to something like IDF_PATH_8266 to differentiate it from that required for ESP32 development?

jhinklejr, 2018-09-28:
Never mind
I worked around it by setting the IDF_PATH in the project make file

weswitt, 2018-09-29:
I think it would be good to have a unique env var for the path. Putting a path into a makefile is not a good idea because the makefile is only good on your computer. I would like to see something like OP suggested.

cgawron, 2018-10-03:
Hm, I hope that there will be one "united" esp-idf in the future and only the compiler toolchain will be different between ESP8266 and ESP32.
I think that is the rationale behind using the same name ...

donghengqaz, 2018-10-09:
We make it use same env value because we may add ESP8266 to esp-idf later.

closed 2018-10-15 by FayeY

micronik-ir, 2022-01-18:
> Never mind I worked around it by setting the IDF_PATH in the project make file

Hello
I have this problem too. How can I fix it?
Thanks