espressif/ESP8266_RTOS_SDK #324: How to obtain the 8266 MAC address through the serial port

invention88, 2018-09-28:
When we are producing the 8266 module, we need to obtain the MAC address to generate the QR code. In the case that the software has no print information output, how to obtain the MAC address of the device through the serial port. Because the software provided by the customer does not print the MAC address from the serial port.

leon-v, 2018-10-02:
This sounds like you are using the built in AT command set.
Is this issue relevant to the RTOS SDK ?

I use this to get the MAC using the V 3.0 of the RTOS SDK...
```
uint8_t mac[6];
esp_efuse_mac_get_default(mac);
sprintf(uniqueID, "%02x%02x%02x%02x%02x%02x", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
```
However i believe there is a different mac for AP and client modes.

FayeY, 2018-10-11:
Could this help? https://github.com/espressif/esptool#read-built-in-mac-address-read_mac

invention88, 2018-10-19:
"esptool.py read_mac" command is available，i test OK， 
thanks.

closed 2018-10-19 by invention88