espressif/ESP8266_RTOS_SDK #325: Making MQTT compative with ipv6

nuclearcat, 2018-09-29:
By default MQTT is not compatible with ipv6 as it uses obsolete gethostbyname in NetworkConnect (and NetworkConnectSSL).
Here is example of same function, but compatible with ipv6 (this particular example forced to ipv6, requesting DNS AAAA query and connecting to resolved v6 address properly on dual stack ESP8266).
It is sligthly different, as it uses instead of integer port - char* pointer to number of port.
Code tested on real network and application.
Not sure if this issue solved in newer PAHO c client library, looking to https://github.com/eclipse/paho.mqtt.embedded-c/blob/master/MQTTClient-C/src/FreeRTOS/MQTTFreeRTOS.c - it seems not.
If you are using that library, i can try to push first updates in their repository.

`	char *address = "hostname";
	char *port = "1883";
	if ((rc = NetworkConnect6(network, address, port)) != 0) { }

static int NetworkConnect6(Network* n, char *addr, char *service) {
	const struct addrinfo hints = {
		.ai_family = AF_INET6,
		.ai_socktype = SOCK_STREAM,
	};
	struct addrinfo *res;
	int err = getaddrinfo(addr, service, &hints, &res);

	if (err != 0 || res == NULL) {
		ESP_LOGE(TAG, "DNS lookup failed err=%d res=%p", err, res);
		vTaskDelay(1000 / portTICK_PERIOD_MS);
		return(-1);
	}
	n->my_socket = socket(res->ai_family, res->ai_socktype, 0);
	if (n->my_socket < 0) {
		ESP_LOGE(TAG, "... Failed to allocate socket.");
		freeaddrinfo(res);
		vTaskDelay(1000 / portTICK_PERIOD_MS);
		return(-2);
	}

	if (connect(n->my_socket, res->ai_addr, res->ai_addrlen) != 0) {
		ESP_LOGE(TAG, "... socket connect failed errno=%d", errno);
		close(n->my_socket);
		freeaddrinfo(res);
		vTaskDelay(4000 / portTICK_PERIOD_MS);
		return(-3);
	}

	ESP_LOGI(TAG, "... connected");
	freeaddrinfo(res);
	return(0);
}
`

nuclearcat, 2018-10-03:
I checked esp-mqtt, and this is what i see there:
`
    static int resolve_dns(const char *host, struct sockaddr_in *ip) {

    struct hostent *he;
    struct in_addr **addr_list;
    he = gethostbyname(host);
`

In SSL it uses mbedtls_net_connect, so maybe mbed are providing proper ipv6 support (i have to check that) and in ssl mode it will work.
So it is not entirely ipv6 ready, at least in some parts.

ustccw, 2019-01-02:
@nuclearcat thanks for your work! we will considerate MQTT based on IPv6 in release v3.2.

shimarin, 2022-04-11:
This may help
https://gist.github.com/shimarin/90a83be3da4d04b4e7f44d7302f310b3

closed 2022-05-10 by nuclearcat