espressif/ESP8266_RTOS_SDK #326: mqtt  发送数据停止

zhuqiyun, 2018-10-01:
mqtt  发送一段时间数据后就停止了 ，看门狗也不起作用。
void user_mqtt_task()
{
    Network network;
    int rc = 0;
    EventBits_t uxBits;

    MQTTPacket_connectData connectData = MQTTPacket_connectData_initializer;

    connectData.clientID.cstring= mqtt_client;
    connectData.willFlag               = 0;
    connectData.MQTTVersion            = 4;
    connectData.keepAliveInterval      = 2;
    connectData.cleansession           = 1;
    connectData.username.cstring= "";
    connectData.password.cstring= "";

    printf("mqtt_client=%s\n",mqtt_client);
    printf("mqtt_user=%s\n",mqtt_user);  
    printf("mqtt_pwd=%s\n",mqtt_pwd);  

    NetworkInit(&network);
    MQTTClientInit(&user_mqtt_client, &network, 2000, mqtt_sendBuf, sizeof(mqtt_sendBuf), mqtt_readBuf, sizeof(mqtt_readBuf));

    Timer timer;
    TimerInit(&timer);

    do
    {
       uxBits = xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT, false, false, portMAX_DELAY);  

    }while((uxBits & CONNECTED_BIT)==0);
        

    xTaskCreate(device_find_task,"device_find_task",2048,NULL,13,NULL);    

    for(;;)
    {
        vTaskDelay(1000 / portTICK_RATE_MS);
        uxBits = xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT, false, false, 1000/portTICK_RATE_MS); 
        printf("----------uxBits=%d\n",(int)uxBits);
        if(uxBits & CONNECTED_BIT) {

            printf("WiFi Connected to ap\n");

        }
        else{     

            continue;
			
        }
        

        if ((rc = NetworkConnect(&network, MQTT_SERVER, MQTT_PORT)) != 0) {
            printf("Return code from network connect is %d\n", rc);
            continue;
        }

        if ((rc = MQTTConnect(&user_mqtt_client, &connectData)) != 0) {
            printf("Return code from MQTT connect is %d\n", rc);
            network.disconnect(&network);            
            continue;
        } else {
            printf("MQTT Connected\n");
        }

        if ((rc = MQTTSubscribe(&user_mqtt_client,topic_down, 2, messageArrived)) != 0) {
            printf("Return code from MQTT subscribe is %d\n", rc);
			MQTTDisconnect(&user_mqtt_client);
            network.disconnect(&network);
            continue;
        } else {
            printf("MQTT subscribe to topic %s\n",topic_down);
        }

        for (;;) {
                 printf("MQTTIsConnected return code=%d\n",rc);
//  sendDataPoint(&dataPoint,&dataPointSW);
                TimerCountdownMS(&timer, 500); /* Don't wait too long if no traffic is incoming */
        #if defined(MQTT_TASK)
                MutexLock(&(user_mqtt_client.mutex));
        #endif
                rc=cycle(&user_mqtt_client, &timer);
               
        #if defined(MQTT_TASK)
                MutexUnlock(&(user_mqtt_client.mutex));
        #endif
               if(rc==FAILURE||rc==7) break;

        }
        MQTTDisconnect(&user_mqtt_client);
        network.disconnect(&network);
        printf("Starting again!rc =========%d",rc);
    }
    printf("mqtt_client_thread going to be deleted\n");
    vTaskDelete(NULL);
    return;
}

发送在另一个线程中

void test_task()
{
    int fd;
    esp_spiffs_deinit(1);
    for(;;)
    {
        vTaskDelay(2000 / portTICK_RATE_MS);
    
        mqttSendData(CMD_TYPE_DATA_PIONT_UP,0,&dataPoint,&dataPointSW);

    }
}

FayeY, 2018-11-13:
想请问，这个问题有更新吗？是否已经解决？

redchenjs, 2018-11-16:
@xuhongv 是只开源了头文件吗

xuhongv, 2018-11-17:
> @xuhongv 是只开源了头文件吗

是的！不过保证，这是我们测试过之后 push 的，绝对稳定，我们有测试数据！我们觉得这个是 Rtos2.0版本最好的MQTT库了！

zhuqiyun, 2018-11-24:
后来放弃了用这个最新的库，继续用的2.x

closed 2019-01-09 by FayeY