espressif/ESP8266_RTOS_SDK #327: Many "undefined" #if variables - can lead to hard to find bugs

jhinklejr, 2018-10-01:
I added  -Wundef to COMMON_WARNING_FLAGS  in project.mk

**************************  my modification
# Warnings-related flags relevant both for C and C++
COMMON_WARNING_FLAGS = -Wundef -Wall -Werror=all \

*******************************************************************

Many of the components have some sort of configuration file with "#include sdkconfig.h" embedded in it to enable the menuconfig process to configure the sdk

Problem is many of the config variables are not defined in sdkconfig.h 

These undefined variables are used in two ways -- you might want to just use one method

First example uses a #ifdef 
#ifdef CONFIG_ESP_LWIP_MEM_DBG

Second example use #if 
#if CONFIG_LWIP_DEBUG

I don't think the #ifdef  method should be used since if it does show up in sdkconfig.h it will be acted upon independent if the config value is a 1 or 0

I recommend the generation of sdkconfig.h that is generated by the menuconfig process be revised to define all config variables - setting default to 0

The current compile process ignores all the undefined warning leading the programmer to think all is well when an issue may be hidden and not identified.

Besides CONFIG variables ... there are multiple other undefines that should also be addressed

Some - but not all

 #if DEBUG
#if __GNU_VISIBLE
#if CONFIG_LWIP_USE_IRAM
 #if LWIP_STATS
#if CONFIG_LWIP_IGMP_DEBUG
#if LWIP_IPV6   ......  note -- ones like this are not defined because the config variable is not
         #define LWIP_IPV6                       CONFIG_LWIP_IPV6
etc .......

Hope that helps

jhinklejr, 2018-10-01:
As I stated in the previous post - in file lwipopts.h you use a mix of #ifdef and #if associated with CONFIG variables from menuconfig

I suggest you only use one -- most likely #if because you can then set 0 or 1 as value

Most all config defines are set using #if 

Here is an example of a multiple define issue I caused ... I'm adding all the CONFIG variables to the end of sdkconfig.h

I missed this set of #ifdef 

#ifdef CONFIG_TCP_OVERSIZE_MSS
#define TCP_OVERSIZE                    TCP_MSS
#endif
#ifdef CONFIG_TCP_OVERSIZE_QUARTER_MSS
#define TCP_OVERSIZE                    (TCP_MSS/4)
#endif
#ifdef CONFIG_TCP_OVERSIZE_DISABLE
#define TCP_OVERSIZE                    0
#endif
#ifndef TCP_OVERSIZE
#error "One of CONFIG_TCP_OVERSIZE_xxx options should be set by sdkconfig"
#endif

CONFIG_TCP_OVERSIZE_MSS = 1 was defined by your menuconfig -- you are setting a value so it should be tested with a #if and not a #ifdef 

I added CONFIG_TCP_OVERSIZE_QUARTER_MSS = 0  in an attempt to define all the CONFIG variables

Issue I caused was that both are defined so the proper value of TCP_OVERSIZE  was not set correctly

I changed all your #ifdef to #if --- and added all undefined CONFIG variables = 0 and I believe I am now getting a more robust and less error prone compulation

jhinklejr, 2018-10-02:
Here is an example of questionable configuration ... its for the ESP32 but it's in this code

const esp_partition_t* esp_ota_get_running_partition(void)
{
    /* Find the flash address of this exact function. By definition that is part
       of the currently running firmware. Then find the enclosing partition. */

#ifdef CONFIG_TARGET_PLATFORM_ESP32
    size_t phys_offs = spi_flash_cache2phys(esp_ota_get_running_partition);

    assert (phys_offs != SPI_FLASH_CACHE2PHYS_FAIL); /* indicates cache2phys lookup is buggy */
#endif

#if CONFIG_TARGET_PLATFORM_ESP32
    esp_partition_iterator_t it = esp_partition_find(ESP_PARTITION_TYPE_APP,
                                                     ESP_PARTITION_SUBTYPE_ANY,
                                                     NULL);
    assert(it != NULL); /* has to be at least one app partition */

    while (it != NULL) {
        const esp_partition_t *p = esp_partition_get(it);
        if (p->address <= phys_offs && p->address + p->size > phys_offs) {
            esp_partition_iterator_release(it);
            return p;
        }
        it = esp_partition_next(it);
    }

    abort(); /* Partition table is invalid or corrupt */
#else
    return esp_ota_get_boot_partition();
#endif
}

The variable CONFIG_TARGET_PLATFORM_ESP32 is used in both a #ifdef and a #if statement

If CONFIG_TARGET_PLATFORM_ESP32 was defined as 0

The #ifdef block but the #if block would not

Is that the expected behavior

Once you start looking for how the CONFIG variables are used -- similar example to this show up a lot

donghengqaz, 2018-10-09:
Thanks a lot, we will take your advice and make the define to be "#if xxx" later and then modify the old ones.

jhinklejr, 2018-10-09:
I've spent hours going over each configuration variable.

What I suggest based on what I've found.

Don't change the method used by lwip and freertos and the ssl libs - they are well established.

For all the one created by Espressif or other developers, I would strongly suggest only using #if - no #ifdef 

By only using #if's - you will have to list every one of the variables in your sdkconfig.h but it will be easy to see exactly how things are configured.

I have found several variables that used #ifdef in some source files and #if for the same variable in other files.  This makes the build dependent on how the compiler reacts to undefined #if statements and difficult to know exactly how the build is constructed.  It also allows you to use an editor with multi-file search capabilities and file config variables that are not really implemented (I sent you an issue on one I found).
As a side note - thanks for providing the capabilities that your IDF is providing.  Getting rid of most of the binary blobs and being able to directly modify source is a major step forward compared to your previous SDK releases.

FayeY, 2018-10-23:
@igrr @projectgus Do you have any comments? It will be better if the ESP8266 RTOS SDK keeps in the same coding style with ESP-IDF. Thanks.

jhinklejr, 2018-10-23:
You can close this out.

 

I’m currently using the IDF as you intended.

 

As I find missing or incorrect variables, I will let you know.

projectgus, 2018-10-23:
Thanks for the update @jhinklejr .

For the record, this coding style comes from  ESP-IDF and there is an (unresolved) request to make this codebase `-Wundef`-clean there as well: https://github.com/espressif/esp-idf/issues/991 - feel free to subscribe to that issue if you'd like to be involved in any further discussions about this.

closed 2018-10-23 by projectgus