espressif/ESP8266_RTOS_SDK #329: Example simple_wifi does not work (probably due to an issue in lwip)

cgawron, 2018-10-03:
## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    4ae1308c2a890920de1896c1e34762068b57d78d
- Development Env:      Make
- Operating System:      Linux
- Power Supply:             USB

## Problem Description

The example simple_wifi does not work. From the log output, the WIFI connection itself seems to work, but the ESP8266 does not get an IP address.

I actually suspect there is an issue in the lwip component, other code (which I have taken from an ESP32 project) also shows that IP connection does not work (neither IPv4 nor IPv6). Basically, I could not get IP connectivity working using the current SDK. 
While trying to track down the issue, I tried the simple_wifi example and noticed that this does not work either. 

The simpe_wife example (taken from the ESP8266 repo, i.e. exactly the same code) works with the ESP32 version of esp-idf (and an ESP32, of course).

### Expected Behavior

Log output should show that the ESP8266 gets an IPv4 address. Here the corresponding output on the ESP32:
´´´
I (5475) I (5475) wifi: state: auth -> assoc (0)
: 
I (5485) I (5485) wifi: state: assoc -> run (10)
: 
I (5515) I (5515) wifi: connected with MCNet, channel 13
: 
I (5525) I (5525) wifi: pm start, type: 1

: 
I (6265) sta ip: 192.168.10.48, mask: 255.255.255.0, gw: 192.168.10.1: 
I (6265) got ip:192.168.10.48: 
´´´

### Actual Behavior

The log ends with the connection to the access point:
´´´
add if0
I (218) simple wifi: wifi_init_sta finished.
I (222) simple wifi: connect to ap SSID:MCNet password:****************
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
pm open phy_2,type:2 0 0
cnt 

connected with MCNet, channel 13
´´´

### Steps to repropduce

1. ``make menuconfig`` (configure STA mode, SSID and Wifi password)
2. ``make -j4``
3. ``make flash monitor``

### Code to reproduce this issue

See [simple_wifi](https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/wifi/simple_wifi) example.  

## Debug Logs

```
11:19 ~/esp8266/simple_wifi $ make flash monitor
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.4.0
Connecting........_____....._
Chip is ESP8266EX
Features: WiFi
MAC: a0:20:a6:00:c1:4f
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 7824 bytes to 5450...
Wrote 7824 bytes (5450 compressed) at 0x00000000 in 0.5 seconds (effective 129.9 kbit/s)...
Hash of data verified.
Compressed 327056 bytes to 211638...
Wrote 327056 bytes (211638 compressed) at 0x00010000 in 18.6 seconds (effective 140.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 83...
Wrote 3072 bytes (83 compressed) at 0x00008000 in 0.0 seconds (effective 2185.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
MONITOR
--- idf_monitor on /dev/ttyUSB0 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x40100000, len 4364, room 16 
0x40100000: _stext at ??:?

tail 12
chksum 0xae
ho 0 tail 12 room 4
load 0x3ffe8008, len 3424, room 12 
tail 4
chksum 0xe0
I (23) boot: ESP-IDF v3.1-dev-188-g4ae1308 2nd stage bootloader
I (23) boot: compile time 11:19:22
I (23) boot: SPI Speed      : 40MHz
I (27) boot: SPI Mode       : QIO
I (31) boot: SPI Flash Size : 2MB
I (35) boot: Partition Table:
I (38) boot: ## Label            Usage          Type ST Offset   Length
I (46) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (53) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (60) boot:  2 factory          factory app      00 00 00010000 000f0000
I (68) boot: End of partition table
I (72) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x48b38 (297784) map
I (139) esp_image: segment 1: paddr=0x00058b50 vaddr=0x3ffe8000 size=0x0052c (  1324) load
I (140) esp_image: segment 2: paddr=0x00059084 vaddr=0x3ffe852c size=0x0012c (   300) load
I (147) esp_image: segment 3: paddr=0x000591b8 vaddr=0x40100000 size=0x06bb4 ( 27572) load
0x40100000: _stext at ??:?

I (161) boot: Loaded app from partition at offset 0x10000
I (6) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (13) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (202) simple wifi: ESP_WIFI_MODE_STA
mode : sta(a0:20:a6:00:c1:4f)
add if0
I (218) simple wifi: wifi_init_sta finished.
I (222) simple wifi: connect to ap SSID:MCNet password:1139653572362251
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
pm open phy_2,type:2 0 0
cnt 

connected with MCNet, channel 13 
```

Junhao-Espressif, 2018-10-09:
Hi @cgawron 

It seems that your AP has some problems.

Could you try another AP, or try portable hot spots with your mobile phones.

Thanks!

FayeY, 2018-10-15:
Closing due to lack of response, please feel free to reopen if this is still an issue and you can supply more details. Thanks.

closed 2018-10-15 by FayeY

mcuwty, 2018-11-09:
@Junhao-Espressif @FayeY 
Hi
I also encountered the same problem in the example simple_wifi. I only modified the WiFi ssid and password in the example. I started the debug level, the last log is displayed as "ERROR dhcp get ip error".
However, using the same router, the esp32 development board works fine in the ESP32 idf .
And if I used ESP8266_NONOS_SDK for my esp8266 development board, it always works fine too.This proves that my router is no problem.
The following is my esp8266 development board print log.

2018-11-09 16:06:23.133:  I (23) boot: compile time 15:19:15
2018-11-09 16:06:23.134:  I (23) boot: SPI Speed      : 40MHz
2018-11-09 16:06:23.142:  I (24) boot: SPI Mode       : QIO
2018-11-09 16:06:23.143:  I (25) boot: SPI Flash Size : 4MB
2018-11-09 16:06:23.143:  I (26) boot: Partition Table:
2018-11-09 16:06:23.143:  I (28) boot: ## Label            Usage          Type ST Offset   Length
2018-11-09 16:06:23.146:  I (30) boot:  0 nvs              WiFi data        01 02 00009000 00004000
2018-11-09 16:06:23.151:  I (32) boot:  1 otadata          OTA data         01 00 0000d000 00002000
2018-11-09 16:06:23.153:  I (35) boot:  2 phy_init         RF data          01 01 0000f000 00001000
2018-11-09 16:06:23.156:  I (37) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
2018-11-09 16:06:23.160:  I (39) boot:  4 ota_1            OTA app          00 11 00110000 000f0000
2018-11-09 16:06:23.163:  I (42) boot: End of partition table
2018-11-09 16:06:23.165:  I (43) boot: No factory image, trying OTA 0
2018-11-09 16:06:23.168:  I (45) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x43d68 (277864) map
2018-11-09 16:06:23.250:  I (103) esp_image: segment 1: paddr=0x00053d80 vaddr=0x3ffe8000 size=0x00ebc (  3772) load
2018-11-09 16:06:23.254:  I (104) esp_image: segment 2: paddr=0x00054c44 vaddr=0x3ffe8ebc size=0x0011c (   284) load
2018-11-09 16:06:23.259:  I (105) esp_image: segment 3: paddr=0x00054d68 vaddr=0x40100000 size=0x0753c ( 30012) load
0x40100000: _stext at ??:?

2018-11-09 16:06:23.270:  I (114) boot: Loaded app from partition at offset 0x10000
2018-11-09 16:06:23.278:  D (121) chip_boot: SPI flash size is 4194304
2018-11-09 16:06:23.280:  
2018-11-09 16:06:23.280:  D (121) chip_boot: SPI flash enable QIO mode
2018-11-09 16:06:23.284:  
2018-11-09 16:06:23.294:  D (4) nvs: nvs_flash_init_custom partition=nvs start=9 count=4
2018-11-09 16:06:23.318:  I (17) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
2018-11-09 16:06:23.324:  D (24) nvs: nvs_open_from_partition backup_mac 0
2018-11-09 16:06:23.330:  D (27) nvs: nvs_get_str_or_blob backup_mac_data
2018-11-09 16:06:23.333:  D (21) nvs: nvs_close 1
2018-11-09 16:06:23.343:  I (34) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
2018-11-09 16:06:23.357:  D (30) nvs: nvs_open_from_partition backup_mac 0
2018-11-09 16:06:23.359:  D (44) nvs: nvs_get_str_or_blob backup_mac_data
2018-11-09 16:06:23.360:  D (48) nvs: nvs_close 2
2018-11-09 16:06:23.364:  D (40) phy_init: loading PHY init data from application binary
2018-11-09 16:06:23.369:  D (54) nvs: nvs_open_from_partition phy 0
2018-11-09 16:06:23.373:  D (57) nvs: nvs_get_str_or_blob cal_data
2018-11-09 16:06:23.379:  D (51) nvs: nvs_get_str_or_blob dc_table
2018-11-09 16:06:23.385:  D (66) nvs: nvs_close 3
2018-11-09 16:06:23.586:  I (269) simple wifi: ESP_WIFI_MODE_STA
2018-11-09 16:06:23.590:  D (262) nvs: nvs_open_from_partition misc 1
2018-11-09 16:06:23.594:  D (276) nvs: nvs_get_str_or_blob log
2018-11-09 16:06:23.597:  D (279) nvs: nvs_open_from_partition nvs.net80211 1
2018-11-09 16:06:23.603:  D (283) nvs: nvs_get opmode 1
2018-11-09 16:06:23.605:  D (286) nvs: nvs_get_str_or_blob sta.ssid
2018-11-09 16:06:23.609:  D (280) nvs: nvs_get_str_or_blob sta.mac
2018-11-09 16:06:23.611:  D (293) nvs: nvs_get sta.authmode 1
2018-11-09 16:06:23.613:  D (296) nvs: nvs_get_str_or_blob sta.pswd
2018-11-09 16:06:23.617:  D (290) nvs: nvs_get_str_or_blob sta.pmk
2018-11-09 16:06:23.621:  D (304) nvs: nvs_get sta.chan 1
2018-11-09 16:06:23.624:  D (307) nvs: nvs_get auto.conn 1
2018-11-09 16:06:23.626:  D (300) nvs: nvs_get bssid.set 1
2018-11-09 16:06:23.630:  D (313) nvs: nvs_get_str_or_blob sta.bssid
2018-11-09 16:06:23.633:  D (317) nvs: nvs_get sta.lis_intval 2
2018-11-09 16:06:23.636:  D (310) nvs: nvs_get sta.phym 1
2018-11-09 16:06:23.639:  D (323) nvs: nvs_get sta.phybw 1
2018-11-09 16:06:23.641:  D (326) nvs: nvs_get_str_or_blob sta.apsw
2018-11-09 16:06:23.645:  D (320) nvs: nvs_get_str_or_blob sta.apinfo
2018-11-09 16:06:23.652:  D (334) nvs: nvs_get sta.scan_method 1
2018-11-09 16:06:23.655:  D (338) nvs: nvs_get sta.sort_method 1
2018-11-09 16:06:23.657:  D (331) nvs: nvs_get sta.minrssi 1
2018-11-09 16:06:23.659:  D (344) nvs: nvs_get sta.minauth 1
2018-11-09 16:06:23.663:  D (347) nvs: nvs_get_str_or_blob ap.ssid
2018-11-09 16:06:23.666:  D (341) nvs: nvs_get_str_or_blob ap.mac
[... 31 lines cut ...]
2018-11-09 16:06:25.584:  D (3747) tcpip_adapter: if0 start ip lost tmr: enter
2018-11-09 16:06:25.586:  D (3741) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x40107ce0 interval=120 ip=0
2018-11-09 16:06:25.593:  D (3751) tcpip_adapter: dhcp client start successfully
2018-11-09 16:06:26.061:  D (4253) tcpip_adapter: dhcpc time(ms): 0
2018-11-09 16:06:26.064:  
2018-11-09 16:06:26.551:  D (4753) tcpip_adapter: dhcpc time(ms): 500
[... cut ...]
2018-11-09 16:06:28.511:  D (6753) tcpip_adapter: dhcpc time(ms): 2500
......
------Ignore similar log here------
......
2018-11-09 16:08:24.814:  D (122759) tcpip_adapter: dhcpc time(ms): 118500
2018-11-09 16:08:24.817:  
2018-11-09 16:08:25.317:  D (123250) tcpip_adapter: dhcpc time(ms): 119000
2018-11-09 16:08:25.318:  
2018-11-09 16:08:25.792:  D (123759) tcpip_adapter: dhcpc time(ms): 119500
2018-11-09 16:08:25.795:  
2018-11-09 16:08:26.282:  D (124259) tcpip_adapter: ERROR dhcp get ip error
2018-11-09 16:08:26.286:

mcuwty, 2018-11-09:
Yes, I switched to the master branch and the problem is solved. But I am using it on my company's products. According to the official advice, I originally used release/v3.0. Should I switch to the master branch?

fmuller-pi, 2019-02-14:
I am using the version v3.1 release branch, and I am still seeing this error `tcpip_adapter: dhcpc time(ms)` randomly where it sometimes end up with a fail if trying too many times...
I know my local network is working fine with an internet connection.
So any recommendation for this? Does the master branch fixes this issue? And if it does, is it safe to move to it knowing that I am using the v3.1 SDK so far?