espressif/ESP8266_RTOS_SDK #333: Memory malloc/free issue inside esf_buf.c which is in your binary blob I think

jhinklejr, 2018-10-09:
## Environment

- Development Kit: ESP8266 V3.0 Release
- IDF version Commit f81c89a  
Head: 4ae1308c2a890920de1896c1e34762068b57d78d
- Development Env:     Make
- Operating System:    Windows using msys32
- Power Supply:         3.3v 2A power supply (ESP8266b embedded in product board)

## Problem Description
ESP8266 RTOS V2 related application software in production over 9 months.
Converted application code to use ESP8266 RTOS V3 to get access to lwip 
Software starts properly and starts a http web server.  Browser attaches and requests first page.
Application issues it's first TCP socket send - to send web page data - and ESP crashes.

Using exception dump (attached) and objdump and map file - I was able to pinpoint the instruction causing the exception  - 0x1c - LoadProhibitedCause.

From my code - the exception occurs after I issue a tcp socket send - the fault occurs in malloc().

I enabled logging in malloc and free just prior to making socket send call -- attached.

The function calling malloc and free several times before the fault is esf_buf.c.  I can not find it in the V3 components so I suspect it is part of a Espressif supplied binary blob.

One thing I noticed in the malloc/free calling sequence prior to the exception is that esf_buf.c make two(2) calls to free buffers that were already freed!!  I consider this a potential issue yet I can't confirm it without the source code.

I'm trying to debug and resolve this issue myself -- do you have any suggestions as to getting access to esf_buf.c?

I have not yet walked line by line through the log file I've attached to see if your implementation of malloc has the issue.

Here is the source for malloc with the exact line causing the issue to fault (one of the pointers was zero -- see log)

This file contains mixed C and assembly and shows line exception occurs at
[Fault Mixed C and Asm.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/2462509/Fault.Mixed.C.and.Asm.txt)

This file contains my debug log prior to the exception.  I noted where socket send is issued and broken the various free and malloc blocks with spaces
[esp.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/2462512/esp.txt)

Is posting here the only way to communicate on an issue?

Thanks in advance for any help or insight.

donghengqaz, 2018-10-10:
Thanks your report and could you supply your code or binary library to us? It may help us to debug the problem.

jhinklejr, 2018-10-10:
Let me see if I can parse it down since its currently in an embedded board being controlled by another cpu.

 

Joe

jhinklejr, 2018-10-10:
I have attached a zip file containing the complete source and binaries to show the issue.  Use link below to download file.

 

http://www.joehinkle.com/download/ESP8266_RTOS_Issue.zip

 

Please don’t make the source of this project public.

 

Please let me know that you have received this email and were able to unzip the folder.

 

In the folder there is a READ_ME file – please read as it explains how to use the code and generate the exception.

 

Thanks in advance for any help you can give on addressing this issue.

 

The folder (not included) Virgin – holds the latest ESP8266_RTOS_IDF_V3 source that the project is build from.

 

It is identified in the IDF_PATH statement in the project’s MakeFile.

 

Joe Hinkle

donghengqaz, 2018-10-11:
I make sure I will not publish the code to others.

I has unzipped the file and build the project, but find:
```
/personal/ESP8266_RTOS_Issue/main/DNS_Client.c:87:19: fatal error: H_dns.h: No such file or directory
 #include "H_dns.h"
```

jhinklejr, 2018-10-11:
Just remove include – I removed all dns

 

Thanks

jhinklejr, 2018-10-11:
I’ve attached it if that’s ok

 

 * @file
 * DNS API
 */

/**
 * lwip DNS resolver header file.

 * Author: Jim Pettinato
 *   April 2007

 * ported from uIP resolv.c Copyright (c) 2002-2003, Adam Dunkels.
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions
 * are met:
 * 1. Redistributions of source code must retain the above copyright
 *    notice, this list of conditions and the following disclaimer.
 * 2. Redistributions in binary form must reproduce the above copyright
 *    notice, this list of conditions and the following disclaimer in the
 *    documentation and/or other materials provided with the distribution.
 * 3. The name of the author may not be used to endorse or promote
 *    products derived from this software without specific prior
 *    written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE AUTHOR ``AS IS'' AND ANY EXPRESS
 * OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED
 * WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
 * ARE DISCLAIMED.  IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR ANY
 * DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
 * DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE
 * GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
 * INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY,
 * WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING
 * NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
 * SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 */

#ifndef LWIP_HDR_DNS_H
#define LWIP_HDR_DNS_H

#include "lwip/opt.h"

#if LWIP_DNS

#include "lwip/ip_addr.h"

#ifdef __cplusplus
extern "C" {
#endif

/** DNS timer period */
#define DNS_TMR_INTERVAL          1000

/* DNS resolve types: */
#define LWIP_DNS_ADDRTYPE_IPV4      0
#define LWIP_DNS_ADDRTYPE_IPV6      1
#define LWIP_DNS_ADDRTYPE_IPV4_IPV6 2 /* try to resolve IPv4 first, try IPv6 if IPv4 fails only */
#define LWIP_DNS_ADDRTYPE_IPV6_IPV4 3 /* try to resolve IPv6 first, try IPv4 if IPv6 fails only */
#if LWIP_IPV4 && LWIP_IPV6
#ifndef LWIP_DNS_ADDRTYPE_DEFAULT
#define LWIP_DNS_ADDRTYPE_DEFAULT   LWIP_DNS_ADDRTYPE_IPV4_IPV6
#endif
#elif LWIP_IPV4
#define LWIP_DNS_ADDRTYPE_DEFAULT   LWIP_DNS_ADDRTYPE_IPV4
#else
#define LWIP_DNS_ADDRTYPE_DEFAULT   LWIP_DNS_ADDRTYPE_IPV6
[... 126 lines cut ...]

/* Multicast DNS definitions */

/** UDP port for multicast DNS queries */
#ifndef DNS_MQUERY_PORT
#define DNS_MQUERY_PORT             5353
#endif

/* IPv4 group for multicast DNS queries: 224.0.0.251 */
#ifndef DNS_MQUERY_IPV4_GROUP_INIT
#define DNS_MQUERY_IPV4_GROUP_INIT  IPADDR4_INIT_BYTES(224,0,0,251)
#endif

/* IPv6 group for multicast DNS queries: FF02::FB */
#ifndef DNS_MQUERY_IPV6_GROUP_INIT
#define DNS_MQUERY_IPV6_GROUP_INIT  IPADDR6_INIT_HOST(0xFF020000,0,0,0xFB)
#endif

#ifdef __cplusplus
}
#endif

#endif /* LWIP_DNS */

#endif /* LWIP_HDR_DNS_H */

jhinklejr, 2018-10-12:
I've implemented my new version of malloc/free that has bad pointer, over-run, and under-run buffer detection and no violations were detected.

The lwip TCP task is the active task when the exception occurs.

As a side note -- since I can only use debug print statements  .... why would the use of est_printf cause task stacks to over-run?

est_printf is the function used in all of your logging functions and it just polls the uart's TX fifo to put another byte in.

I can see the usage slowing things down but why is it eating up stack space?

Just engage logging in your malloc/free routines and you will get a stack over-run exception very soon.

THAT should not happen.

Any thoughts?

jhinklejr, 2018-10-12:
I think I found the issue.

I decided to start tracking down lwip send() since that that is the function causing the exception.

send is replaced via a define with lwip_send

lwip_send is replaced with lwip_send_esp for multi-task sockets.

Look at the implementation for lwip_send_esp:

int lwip_send(int s, const void *data, size_t size, int flags)
{
    int ret;

    SOCK_MT_ENTER_CHECK(s, SOCK_MT_LOCK_SEND, SOCK_MT_STATE_SEND);

    ret = lwip_send_esp(s, data, size, flags);

    SOCK_MT_EXIT_CHECK(s, SOCK_MT_LOCK_SEND, SOCK_MT_STATE_SEND);

    return ret;
}

The function calls itself - like a dog chasing its tail --- round and round.

THAT is not the only cross-linked socket call -- looks to be a few.

Not quite sure right now what SEND function is actually the one to be called -- not send, not lwip_send, and not lwip_send_esp because the are all the same.

Please let me know.

Thanks

jhinklejr, 2018-10-12:
Please close this issue out.

donghengqaz, 2018-10-15:
Year， I have found that "ets_printf" has some problems, but its a rom code and has no method to modify it, I have tried use outside function to replace it.

closed 2018-10-15 by donghengqaz