espressif/ESP8266_RTOS_SDK #334: I think you have esp_heap_cap.c bug - specifically blk_link_size

jhinklejr, 2018-10-10:
I issued concern #333 and was attempting to validate your heap functions since issue #333 is a heap related issue.

I took source for esp_heap_cap.c and their header and made a Windows program that would allow me to execute the functions in a good debugging environment.

I just started with function void esp_heap_caps_init_region(heap_region_t *region, size_t max_num)
and think I found an issue -- i'll explain but please tell me if i'm wrong.

Variable region (assume only one region right now) is initialized with the starting address of the heap memory and the number of bytes associated with that space.

The Heap Management uses a structure to define how memory is used with that space.

mem_blk_t *mem_start, and *mem_end are positioned within the heap space to identify the start of available memory and the end of available memory

mem_blk_t  is defined as:
typedef struct mem_blk 
{
    struct mem_blk  *prev;  ///< Point to previous memory block
    struct mem_blk  *next;  ///< Point to next memory block 
} mem_blk_t;

I could not find any documentation on your heap implementation so I've had to do some reverse engineering to make sure I understand what the code is actually doing

The actual init function follows

void esp_heap_caps_init_region(heap_region_t *region, size_t max_num)
{
    uint8_t num;
    mem_blk_t *mem_start, *mem_end;

    for (num = 0; num < max_num; num++) 
	{
        mem_start = (mem_blk_t *)HEAP_ALIGN(region[num].start_addr);
        mem_end = (mem_blk_t *)(HEAP_ALIGN(region[num].start_addr + region[num].total_size));
        if ((uint8_t *)mem_end != region[num].start_addr + region[num].total_size)
            mem_end = (mem_blk_t *)((uint8_t *)mem_end - sizeof(byte*));
        mem_end = (mem_blk_t *)((uint8_t *)mem_end - MEM_HEAD_SIZE);	// less size of one block type used

        mem_start->prev = NULL;
        mem_start->next = mem_end;

        mem_end->prev = mem_start;
        mem_end->next = NULL;

        g_heap_region[num].free_blk = (byte*)mem_start;
        g_heap_region[num].min_free_bytes = g_heap_region[num].free_bytes = blk_link_size(mem_start);
    }
}

My problem is with the last line as I'll explain

In my simulation (Windows program) I established a byte array for the heap and assigned a size to the array of 5075 -- I wanted to make sure it was not an multiple of 4 so I could make sure your HEAP_ALIGN was working properly.

My heap memory made available to the heap management was from address 64e0 to 78b2 - which is 

at the end of the function:

mem_start (p:0, n:78a8)   .... Im stating the actual values for the previous and next pointers.
mem_end (p:64e0, n:0)

&mem_start  = 64e0
&mem_end = 78a8 ... start if struct positioned 10 bytes with the end of allocated memory to achieve alignment and compensate for the size of the structure

mem_start's address is the very first properly aligned memory location at the start of the heap space.
mem_end's address locates it at the end of the heap less the size of the structure so that the mem_end structure is within the heap space.

Here is my concern -- I suspect your code is wrong....

line:
g_heap_region[num].min_free_bytes = g_heap_region[num].free_bytes = blk_link_size(mem_start);

assigns a vale for "free bytes" --- I take that as how many free bytes are available for allocation.

the issue is the expansion of "blk_link_size" which appears to WANT to provide the number of bytes available within THAT block for possible allocation.

blk_link_size is also used within the malloc code so that is why I'm pointing this out.

blk_link_size (after you expand all the macros) takes the "next" pointer and subtracts the address of the block.  So as the code is written -- it expands to

78a8 - 64e0 = 0x13c8 or 5064 decimal

My concern is that the value of 5064 is NOT correct as it also includes the size of the mem_start block (8 bytes).  I believe the correct value should be 5064 - 8 = 5056 or 0x1c30.

I think blk_link_size  is over stating available memory which can lead to memory corruption issues.

I thin the proper implementation for blk_link_size  should be:

currently:

size_t blk_link_size(mem_blk_t *blk)
{
    return (size_t)mem_blk_next(blk) - (size_t)blk;
}

suggested change

size_t blk_link_size(mem_blk_t *blk)
{
    return (size_t)mem_blk_next(blk) - (size_t)blk - sizeof(mem_blk_t);
}

The proposed changes removes the size of the structure from the available memory.

Please let me know what you think.  Am I correct?

Thanks.

jhinklejr, 2018-10-11:
I was wrong in that blk_link_size should NOT be changed!!!  It's providing a memory size that includes the smallest struct size - (if tracing is enabled as it is for me) the struct grows larger to include file name and line number.

So - my previous post turns out to be more of a learning experience as the code is fine.  Sorry for issuing the post.  I'm still simulating malloc to make sure all is well.

donghengqaz, 2018-10-11:
Thanks for your simulating, you will make us do better. :)

closed 2018-10-11 by FayeY