espressif/ESP8266_RTOS_SDK #335: Malloc/Free Trace concern

jhinklejr, 2018-10-11:
I posting this here because if a question is asked about ESP8266_RTOS V3 or IDF - it never gets answered.

I'm reviewing malloc/free implementation used in the V3.0 RTOS SDF.

It has a "Trace" capability that can be turned on/off by the user making a call.

I think the Trace capability is very worth while - I think the way it is implemented is incorrect.

To fix my concern (I'll discuss below) - remove the option to turn Trace off and on within the application and just make it a BUILD option -- either always on or always off --- here's why.

Malloc uses a HEAD structure (actually a previous pointer and a next pointer) to allocate and free the memory block.

For a block being allocated via malloc - bit 0x80000000 is set in the next pointer and if Trace is active, bit 
0x80000000 is set in the previous pointer.

A different HEAD block structure is used if Trace is active or not.

Trace NOT active - HEAD is just the next and prev pointers.

if Trace is active - the the structure is 8 bytes larger holding a pointer to a string identifying which file the malloc call was from and also what the caller's line number within the file.  These 8 bytes follow the next/prev pointers.

This is all well in malloc because malloc has a pointer to the HEAD block associated with the memory pointer it is going to return.

The problem comes in when free is called and is passed the pointer to the memory that was allocated by malloc.  Free must do something to convert the incoming memory pointer into a pointer to its HEAD structure.

This is the line of code in free that performs that task.

    mem_blk = ptr2blk(ptr, ptr_is_traced(ptr));

If we expand ptr_is_traced() - it's job is to tell use if this memory block was allocated with Trace or not.

Here is the code associated with that function -- I believe there are two(3) issues associated with it.

static inline bool ptr_is_traced(void *ptr)
{
    uint32_t *p = (uint32_t *)ptr - 1;

    return p[0] & 0xf0000000 ? false : true;
}

ptr_is_traced gets the 4 bytes of memory just in front of the data pointer and assumes IT holds the TRACE flag.

Issue 1: -- that assumption is valid on a HEAD struct that was not created when Trace was active because the 4 bytes in front of the data pointer is the previous pointer -- the one we set in malloc.  If Trace was active -- the 4 bytes in front of the data pointer holds the line number!!! -- and has no Trace flag associated with it -- SO -- if Trace was active -- ptr_is_traced will not use the correct memory to make the determination.

Issue 2:  To me - this is just bad practice.  Bit 0x80000000 was used in malloc to identify Trace yet the complete upper nibble is used in Free -- this can cause problems if the other bits of the nibble are set outside of bit 0x80000000.

Issue 3:  The wrong bool flag is being returned based on the comparison.  Currently - false is being returned if the upper nibble is non-zero --- I think true should be returned.

Currently, the TRACE capability has the ability to turn trace off and on at will of the application.  The problem with this is that FREE has no idea if the HEAD it is working with is a 8 byte head (no trace) or a 16 byte head (trace active).

I suggest that TRACE be a build ON/OFF option and the malloc/free code be modified based on the build option to always uses the correct memory location in the HEAD block (FREE will always know if trace is active or not so no determination is necessary - the mem_blk will always be a fixed number of bytes ahead of the data pointer.

Let me know if you agree.

Thanks.

jhinklejr, 2018-10-11:
The first line above was supposed to be:
I posting this here because if a question is asked about ESP8266_RTOS V3 or IDF  on the **bbs.Espressif.com site -**-- - it never gets answered.

donghengqaz, 2018-10-12:
Thanks for your point and I may fix this immediately. Later I will add macro to enable or disable the memory trace.

donghengqaz, 2018-10-12:
I add this to make it easy to use and user can use this at the last release code or only debug version.

donghengqaz, 2018-10-12:
I will continuously to fix the bug according to your point.

donghengqaz, 2018-10-12:
Others manage ```bbs.Espressif.com site```, so you can add issues here. I will see these issues everyday as possible and talk to you.

closed 2018-10-15 by wujiangang