espressif/ESP8266_RTOS_SDK #339: C++ linkage error _impure_ptr

THKDev, 2018-10-15:
## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
c4f2a58351b362640f7eb9bf8df147e83970b50c
- Development Env:      Make

## Problem Description
toolchains/xtensa-lx106-elf-4.8.5/bin/../lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/lib/libstdc++.a(vterminate.o):(.literal._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x8): undefined reference to `_impure_ptr'
collect2: error: ld returned 1 exit status

### Expected Behavior
Should link without error.

donghengqaz, 2018-10-16:
Thanks for your report, I think maybe adding the C++ library and header files to SDK is OK like newlib, and I will report the bug to the toolchain developer.

r44083, 2018-10-23:
Hi, I think that this issue could be related to this one: #283 
I created it THREE month ago and there is no any solution!

donghengqaz, 2018-10-26:
@THKDev I am recompiling the crosstool chian to resole the problem and will publish it later.

FayeY, 2018-12-20:
This issue should have been solved in the latest SDK. Please have a try. Thanks.

closed 2018-12-20 by FayeY