espressif/ESP8266_RTOS_SDK #340: LWIP - esp8266 port wrong format identifier

jhinklejr, 2018-10-15:
In your file for the port ---

file include\arch\cc.h you have the following format identifiers stated

#define S16_F "d"
#define U16_F "d"   ----- wrong
#define X16_F "x"

#define S32_F "d"
#define U32_F "d"  ---- wrong
#define X32_F "x"

They should be

#define S16_F "d"
#define U16_F "u"
#define X16_F "x"

#define S32_F "d"
#define U32_F "u"
#define X32_F "x"

donghengqaz, 2018-10-16:
Thanks for your help and I will apply the fixes.

donghengqaz, 2018-10-16:
Maybe we can use following:
```
#define S16_F "hd"
#define U16_F "hu"
#define X16_F "hx"
```

jhinklejr, 2018-10-16:
Dongheng

 

My only comment is that the hd format identifier is not one used by small compact printf routines for embedded systems – maybe big machines.

 

All printf argument variables that are push to the stack on the esp8266 are all 32 bits wide and if the compiler did its job noting that the actual variable was 16 bits – it would have set the upper word accordingly.

 

 

Thanks for looking into this

closed 2018-10-16 by wujiangang