espressif/ESP8266_RTOS_SDK #342: Explain your use of Socket_MT

jhinklejr, 2018-10-16:
lwip and your documentation is vague on multithreading.

From my reading - the use of a socket in lwip is restricted to a single thread but multiple threads can create and use a socket as long as socket activity is restricted to that thread.

If THAT is true - then what is the purpose of your socket_mt implementation?

Is your implementation ONLY required is a specific socket is to be used by multiple threads and your lock/unlock of that socket makes sure one thread's request is returned before another thread can have access?

Here is why I ask ... I have a thread that only works with a single TCP socket.  I have another thread that works with multiple UDP sockets.

Do I really need to use your socket_mt or an I thread safe without it?

Thanks

donghengqaz, 2018-10-17:
I think you can use "socket_mt" to make socket mutil-thread safe, because we internal testcases have cover some multi-thread case, it may reduce some cpu usage, but a little safer.

donghengqaz, 2018-10-17:
If your function will not make these tasks using sockets competition and synchronization， you can drop this "mt" function.

jhinklejr, 2018-10-24:
Please close

closed 2018-10-25 by FayeY