espressif/ESP8266_RTOS_SDK #344: UDP Send Sync - What is it

jhinklejr, 2018-10-16:
In the config udp send sync can be enabled.

I can't figure out from looking at udp_sync.c why/where this would be used.

Please explain what it does and why I might want to enable it.

Thanks

donghengqaz, 2018-10-17:
The function just server for our extreme applications:
```
while(1) send_udp;
``` 
we add this to let upper-low function which send UDP packet block until low-level sends it over(not OK), and it can reduce unnecessary CPU overhead and increased UDP sending success rate.

jhinklejr, 2018-10-19:
Can you please explain a little more?

I have seen where I am sending a UDP msg to another device that there can be a big time delay in the other device receiving it.  I did not know if it was a xmit delay or a delay in the other device receiving the msg and delaying in posting it to the application.

Does this help remove time delay in xmit operations.

You stated "server" --- it is just server based operations?

donghengqaz, 2018-10-24:
It is that lowlevel(wifi) don't care if the packet has been passed to hardware send buffer, so we add the function to make sure the UDP packet has been pass there then return the OK.

jhinklejr, 2018-10-24:
Please close

closed 2018-10-25 by FayeY