espressif/ESP8266_RTOS_SDK #349: release/v2.x.x erase flash sector 1

berkutta, 2018-10-19:
I'm trying to update the Bootloader in SW. For that I need to erase Sectors which seem to have a some sort of protection. 

I'm using the following code:
`printf("Starting to do SPI Flash stuff \n");

printf("Disabling SPI erase protection \n");

spi_flash_erase_protect_disable();

printf("Erasing SPI Flash sector 0 \n");

spi_flash_erase_sector(0);

printf("Erasing SPI Flash sector 1 \n");

spi_flash_erase_sector(1);

printf("Erasing SPI Flash sector 2 \n");

spi_flash_erase_sector(2);

printf("Finished with SPI Flash stuff \n");`

The output is the following:
`Starting to do SPI Flash stuff 
Disabling SPI erase protection 
Erasing SPI Flash sector 0 
Erasing SPI Flash sector 1 
�
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
ets_main.c 
`

Is there any way other than spi_flash_erase_protect_disable to disable this Flash protection and allowing me to erase Sector 1? (Yes, I know it's dangerous)

berkutta, 2018-10-23:
You're right, thought bootloader is bigger 🙃

berkutta, 2018-10-23:
@27-301 Ok, second thing. After updating the bootloader (while ensuring to be on userbin1) I can't get system_restart() to work. I think it restarts at the userbin location or something like that and just outputs jibberish. Is there anything possible that I can to a proper reset? If I push the reset button on my dev board it boots perfectly from the new Bootloader.

berkutta, 2018-10-26:
If I do a "reset" through deep sleep I can perfectly boot the new Bootloader. But unfortunately my production hardware doesn't have the Connection between GPIO16-RESET. So is there any Workaround to do a similiar reset like via the GPIO16-RESET wire in pure Software? @27-301

27-301, 2019-01-16:
if you use deep sleep, by default it will be trigged by time, gpio16 is MUST.

close firstly, if have other issues, please raise new, thanks.

closed 2019-01-16 by 27-301