espressif/ESP8266_RTOS_SDK #351: RTSO SDK V3.1 ADC

pengu-lee, 2018-10-23:
Is there any recommended method for ADC reading function, like the function system_adc_read() in NONOS SDK?

pc0808f, 2018-11-02:
I am waiting this ADC function, too.

x3ro, 2018-11-04:
Fascinatingly enough, it seems to exist, at least in RTOS SDK v3.0. I'm at revision `39e5551` and `components/esp8266/lib/libcore.a` contains the relevant symbols:

~~~
ψ xtensa-lx106-elf-nm /Users/lucas/_dev/esp8266/ESP8266_RTOS_SDK/components/esp8266/lib/libcore.a  | grep adc
xtensa-lx106-elf-nm: efuse.o: no symbols
         U phy_adc_read_fast
00000178 T system_adc_read
000001c8 T system_adc_read_fast
~~~

Using said SDK version, the following works for me:

~~~
#include <stdio.h>
#include <inttypes.h>

#include "FreeRTOS.h"
#include "task.h"
#include "esp_system.h"

extern uint16_t system_adc_read(void);

void app_main(void)
{
	printf("SDK version:%s\n", esp_get_idf_version());
	while(true) {
		uint16_t res = system_adc_read();
		printf("res: %" PRIu16 "\n", res);
		vTaskDelay( pdMS_TO_TICKS( 500) );
	}
    
}
~~~

and results in 

~~~
I (32) boot: ESP-IDF v3.0 2nd stage bootloader
[...]
SDK version:v3.0
res: 1024
res: 924
res: 921
res: 918
~~~

I've no idea why the SDK doesn't expose the functions, maybe they're not working correctly yet?

pc0808f, 2018-11-27:
I try to use code form @x3ro  and it is work now~

SDK version:v3.1-dev-359-gcd25009
res: 825
res: 825
res: 824
res: 824
res: 825

closed 2019-01-21 by donghengqaz

x3ro, 2019-01-21:
Was this closed because it is fixed?

27-301, 2019-01-21:
@x3ro This is supported in ed02e834ac8afab2b2d9f2bf8031049234fb162a, now we have a new API: `uint16_t adc_read();`, you can check it in `components/esp8266/include/driver/adc.h`.

Thanks

x3ro, 2019-01-22:
Cool, thanks!