espressif/ESP8266_RTOS_SDK #355: The automatic reconnection of the WiFi connection no longer works

THKDev, 2018-10-24:
## Environment

- Development Kit: ESP-12E
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
d28329d7139673431efccff95edb173212dab259
- Development Env:      Make
- Operating System:     Linux
- Power Supply:         external 3.3V

## Problem Description
Automatic reconnect of WiFi connection not working anymore. 

### Expected Behavior
Automatic reconnect of WiFi as it used to be.

### Actual Behavior
No reconnect.

### Steps to repropduce
1. Connect ESP8266 to AP
2. Switch AP off
3. wait 30 seconds
4. Switch AP on again
## Debug Logs
```
2018-10-24 21:23:36.354:  bcn_timout,ap_probe_send_start
2018-10-24 21:23:38.854:  ap_probe_send_over, rest wifi status to disassoc
2018-10-24 21:23:38.859:  state: 5 -> 0 (1)
2018-10-24 21:23:38.863:  rm 0
2018-10-24 21:23:38.863:  pm close 7 0 0/25080329
2018-10-24 21:23:38.867:  D (29922) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:SSID, ssid_len:4, bssid:71:4d:7b:4d:32:68, reason:200                                     
2018-10-24 21:23:38.880:  W (29930) mqttConnectCallback: mqtt connect failed or broken.                                                                                  
2018-10-24 21:23:38.889:  D (29938) eQueueSendWithData: Send message 7
2018-10-24 21:23:38.893:  D (29944) tcpip_adapter: if0 start ip lost tmr: enter
2018-10-24 21:23:38.901:  D (29951) tcpip_adapter: if0 start ip lost tmr: interval=120                                                                                   
2018-10-24 21:23:38.910:  I (29959) eventLoopCb: Got new state 5
2018-10-24 21:23:38.914:  I (29964) eventLoopCb: SYSTEM_EVENT_STA_DISCONNECTED
2018-10-24 21:23:38.923:  D (29972) eQueueSendWithData: Send message 2
2018-10-24 21:23:38.927:  D (29978) eMessageHandlerTask: eMessage received: 7
2018-10-24 21:23:38.936:  D (29985) eMessageHandlerTask: eMessage received: 2
2018-10-24 21:24:09.500:  I (60563) observerTask: Date: Thu Jan  1 00:00:59 1970, minheap: 48552, heap: 54816                                                            
2018-10-24 21:25:09.506:  I (120566) observerTask: Date: Thu Jan  1 00:01:59 1970, minheap: 48552, heap: 54584                                                           
2018-10-24 21:25:38.889:  D (149950) tcpip_adapter: if0 ip lost tmr: enter
2018-10-24 21:25:38.894:  D (149953) tcpip_adapter: if0 ip lost tmr: raise ip lost event                                                                                 
2018-10-24 21:25:38.902:  D (149956) event: SYSTEM_EVENT_STA_LOST_IP
2018-10-24 21:25:38.906:  I (149958) event: station ip lost
2018-10-24 21:25:38.910:  I (149961) eventLoopCb: Got new state 8
```

27-301, 2018-10-25:
You can handle the reconnection in SYSTEM_EVENT_STA_DISCONNECTED event.

THKDev, 2018-10-25:
@27-301 
I tried this but NO connection will be established. Could you provide some working code for this case please.

ustccw, 2018-10-25:
@THKDev have you tried this way:
```
case SYSTEM_EVENT_STA_DISCONNECTED:
        esp_wifi_connect();
        break;
```

closed 2018-10-26 by THKDev