espressif/ESP8266_RTOS_SDK #356: GPIO ouptut to input readjusting has a lot delay

nolex90, 2018-10-25:
- Development Kit: ESP201
- IDF version: d28329d
- Development Env:    make, VScode
- Operating System:    Ubuntu
- Power Supply:        external 3.3V

## Problem Description

I am implementing a DHT22 driver, and I experienced that if I readjust GPIO output to input, there is a 7ms delay. 

### Expected Behavior

In my opinion, the 7ms delay is huge, comparing to other embedded socs.
For the driver implementation, I know that I can use input and toggling the pullup resistor to generate 1-0 transaction. I posted this issue to investigate whether it is a bug.

### Actual Behavior

See picture.

### Steps to repropduce

Set GPIO output, and set to 1.
Set GPIO to input, with ISR, and see the pins state.

### Code to reproduce this issue
https://gist.github.com/nolex90/c5c711f777d7dd1229298dc3a83828d7
Result:
[image]

xiongyumail, 2018-10-25:
Hi, @nolex90 

When gpio is initialized, some log information will be printed out through the serial port first. So make you think the configuration is slower.

Once you have initialized gpio, you can use the following function to configure gpio directly into input mode.
gpio_output_disable(gpio_num_t gpio_num)；

nolex90, 2018-10-25:
Such a noobie problem.

Yes, as you said the UART sending messages which provided by the library, it takes times. I solved with this 2 line modification:
	gpio_set_direction(DHT_PIN, GPIO_MODE_DEF_INPUT);
	gpio_set_intr_type(DHT_PIN,GPIO_INTR_POSEDGE);

closing issue.

closed 2018-10-25 by nolex90