espressif/ESP8266_RTOS_SDK #357: gen_appbin.py script is not working with Python 3

h1aji, 2018-10-27:
Modern distribution does not use python 2 these days.
gen_appbin.py should be able running on python 3.

When I am trying to, I see this error:

> TypeError: a bytes-like object is required, not 'str'

And the issue is in this part:

```

def write_file(file_name,data):
	if file_name is None:
		print 'file_name cannot be none\n'
		sys.exit(0)

	fp = open(file_name,'ab')

	if fp:
		fp.seek(0,os.SEEK_END)
		fp.write(data)
		fp.close()
	else:
		print '%s write fail\n'%(file_name)
```

donghengqaz, 2018-10-30:
The print cause it, you can remove the line.

h1aji, 2018-10-30:
Its not print, its fp.write(data)

```
def write_file(file_name, data):
    if file_name is None:
        print ('file_name cannot be none\n')
        sys.exit(0x0)

    with open(file_name, 'ab') as fp:
        if fp:
            fp.seek(0x0, os.SEEK_END)
            fp.write(data)
            fp.close()
        else:
            print ('%s write fail\n' % file_name)
```

This is python 3 code

h1aji, 2018-11-11:
Any update on this?

h1aji, 2018-11-21:
@donghengqaz Can you please help with this, it looks small issue. But I am unable to resolve it. "data" comes as a string and when I try open file without "b" attribute I get message says that its byte-like object.
I dont understand what kind of type is "data". Would you be able to assist? Thanks in advance

FayeY, 2018-12-20:
Hi, so sorry that python2 may be a better choice, it is the version we are using. Maybe we will try python 3 in the future. Thanks.

closed 2018-12-20 by FayeY