espressif/ESP8266_RTOS_SDK #359: ESP8266 V3 - I want to implement a different stack

jhinklejr, 2018-10-28:
I have a lot of experience with the FreeRtosTCP stack and would like to see if I can use it instead of the lwip stack provided.

I have a couple of questions toward that end:

1.  Do any of your binary blobs depend directly on lwip or do they just expect buffers to be passed to them?

2.  If they just expect buffers - can you provide me with the function called when data is received and function to call when a buffer is to be transmitted.

I know these questions seem simplistic - but if I can get access to where your binary blobs interface with the "stack" then I would like to see if I can port a new stack in.

I wrote a driver for the FreeRtosRTCP using a Kinetis ARM cpu so I'm familiar with porting.  I find I prefer the FreeRtosTCP stack over the lwip -- hence my question.

If I can port it - I will share for both the ESP8266 and the ESP32 IDF.

Thanks.

jhinklejr, 2018-10-28:
I'm not suggesting this port would be something you folks are expected to support.

I initially want it for myself.

jhinklejr, 2018-10-28:
Does you wifi driver pass incoming packets via a queue?

jhinklejr, 2018-10-29:
Please verify that file esp_socket.c is the proper interface between IDF code (non-binary blob) and any stack to be implemented.

donghengqaz, 2018-10-30:
We have removed LWIP data from internal WIFI library, so you may just add your own tcpip adapter layer based on your TCP/IP stack. If there is something problem, you can ask here.

jhinklejr, 2018-10-31:
Heads up ... the following 3 variables are allocated in the lwip component folder AND are used within your binary blob.

I had to declare them in my code to satisfy linking.

You did not make a clean break from lwip as stated in your post above.

const struct eth_addr ethbroadcast = {{0xff,0xff,0xff,0xff,0xff,0xff}};
const struct eth_addr ethzero = {{0,0,0,0,0,0}};
unsigned long LwipTimOutLim = 0;

If you really want a clean break from lwip with no bread crumbs left behind - these 3 variables need to be addressed.

Thanks.

donghengqaz, 2018-10-31:
There is some history problem, so we have not removed all clearly.
`ethbroadcast` and `ethzero` will be cleaned later.

`LwipTimOutLim ` This not LWIP private data, this is for library power manage code. We may refactor this component later.

Thanks for your advise.

jhinklejr, 2018-10-31:
Thanks.

Please close

closed 2018-11-01 by FayeY