espressif/ESP8266_RTOS_SDK #361: 使用EXTRA_COMPONENT_DIRS添加自定义组件，make后自定义组件不会编译

ChampOnBon, 2018-10-29:
我在project_template文件夹里面的Makefile文件中添加EXTRA_COMPONENT_DIRS调用外部自定义组件，内容如下所示：
PROJECT_NAME := project_template
include $(IDF_PATH)/make/project.mk
EXTRA_COMPONENT_DIRS += $(IDF_PATH)/chp_lib/chp_iot
EXTRA_COMPONENT_DIRS += $(IDF_PATH)/chp_lib/gal
EXTRA_COMPONENT_DIRS += $(IDF_PATH)/chp_lib/osil
在编译的时候，三个自定义组件文件夹不会通过编译，而如果把三个文件夹添加到IDF_PATH下的components文件夹中，可以正常通过编译，是不是EXTRA_COMPONENT_DIRS的使用方法有问题？

ustccw, 2018-10-29:
@ChampOnBon 可以试一下绝对路径，类似下面这种：
```
EXTRA_COMPONENT_DIRS := $(realpath ../../)
```

donghengqaz, 2018-10-30:
```
EXTRA_COMPONENT_DIRS += $(IDF_PATH)/chp_lib/chp_iot
EXTRA_COMPONENT_DIRS += $(IDF_PATH)/chp_lib/gal
EXTRA_COMPONENT_DIRS += $(IDF_PATH)/chp_lib/osil
```
加在
```
include $(IDF_PATH)/make/project.mk
```
上面

closed 2018-11-02 by ChampOnBon