espressif/ESP8266_RTOS_SDK #363: How determine link status on AP and on STA interfaces - dhcp client

jhinklejr, 2018-10-31:
I am porting FreeRTOS_TCP stack and replacing lwip.

I see you have started this on the ESP32 and you have posted a very simplistic network_Interface (only STA mode).

I need to be able to get the "link status" of both interfaces so I can inform the stack.

Should I always consider the AP link status UP after AP mode is set and config and esp_wifi_start() called or is there an event or CallBack that can be used?  The ESP32 network_interface code suggests their is a callback or event.

Same question on STA - once STA mode is set and config  and esp_wifi_start() called - how can I get link status?

When STA is activated with esp_wifi_start() - does the stack (currently lwip or FreeRTOS_TCP) use their dhcp client to acquire an IP from the access point or do you do that within the wifi binary blob and NOT involve the stack.  If that is true I can watch for a STA_IP event.

Thanks in advance for you reply.

donghengqaz, 2018-10-31:
I do not join adding FreeRTOS_TCP to esp32, so I just know little. You can use event callback "esp_event_loop_init(event_handler, NULL)" to get the status of WIFI link.

jhinklejr, 2018-10-31:
Q1:  Is event SYSTEM_EVENT_AP_START used to identify AP link up?

Q2:  Is event SYSTEM_EVENT_STA_CONNECTED used to identify STA link up?

Q3:  In STA mode - is lwip's DHCP clint used to acquire STA IP or do you use your own binary blob version?

jhinklejr, 2018-10-31:
I think I found all my answers by reviewing event_default_handlers.c

jhinklejr, 2018-10-31:
Please close

closed 2018-11-01 by FayeY