espressif/ESP8266_RTOS_SDK #364: What are the Proper functions to send and receive wifi  completely formed Ethernet packets

jhinklejr, 2018-10-31:
I am removing lwip and porting in a new stack.

The functions in file tcpip_adapter appear to be the main interface for incoming packets and controlling the DHCP server.

Is function static int tcpip_adapter_recv_cb(struct esp_aio *aio) called by your binary blob when a packet is available?

I can't find the send function.

Is function int esp_aio_sendto(esp_aio_t *aio, const struct sockaddr_ll *to, socklen_t len) the one that should be used?

Thanks

donghengqaz, 2018-10-31:
First you should create, bind the low-level network card to your index and register the receive callbakc function like:
```
static int tcpip_adapter_bind_netcard(const char *name, struct netif *netif)
{
    int s, ret;
    s = esp_socket(AF_PACKET, SOCK_RAW, ETH_P_ALL);
    if (s < 0) {
        ESP_LOGE(TAG,"create socket of (AF_PACKET, SOCK_RAW, ETH_P_ALL) error\n");
        return -1;
    }

    ret = esp_ioctl(s, SIOCGIFINDEX, name);
    if (ret) {
        ESP_LOGE(TAG,"bind socket %d to netcard %s error\n", s, name);
        esp_close(s);
        return -1;
    }

    ret = esp_aio_event(s, ESP_SOCKET_RECV_EVENT, tcpip_adapter_recv_cb, netif);
    if (ret) {
        ESP_LOGE(TAG,"socket %d register receive callback function %p error\n", s, tcpip_adapter_recv_cb);
        esp_close(s);
        return -1;
    }
    return s;
}
```
Network card will post the Ethernet packet here.

And the you can call  function to send packet with the send over(not send OK) callback function as following code:
```
    esp_aio_t aio;

    aio.fd = (int)netif->state;
    aio.pbuf = p->payload;
    aio.len = p->len;
    aio.cb = low_level_send_cb;
    aio.arg = p;
    aio.ret = 0;

    err = esp_aio_sendto(&aio, NULL, 0);
```

jhinklejr, 2018-10-31:
In the post above you stated to register the recv callback using function:

 ret = esp_aio_event(s, ESP_SOCKET_RECV_EVENT, tcpip_adapter_recv_cb, netif);

I won't be using a struct netif in my stack.

I don't have access to the code in esp_aio_event() so can you tell me exactly what fields within netif is used by this function.  I can then use a dummy netif with the required data.

jhinklejr, 2018-10-31:
Never mind.  Spoke too early.  I do have access and it's just an argument passed back with the callback.

jhinklejr, 2018-10-31:
I think it's the associated socket for the interface - correct?

That is what appears to be passed to netif_add() which is used to set the initial state value.

donghengqaz, 2018-11-01:
`aio.fd` is the target file description you call `esp_socket(AF_PACKET, SOCK_RAW, ETH_P_ALL)` return.

Q1. It is called at task not interrupt

Q2. Yes， so you should call `heap_caps_malloc(n, MALLOC_CAP_8BIT)` to malloc the payload.

jhinklejr, 2018-11-02:
I'm confused.

My question #2 asked if the buffer sent to esp_aio_sendto() needed to be aligned.  I had asked 4 byte alignment as that is what most DMA wants.

The reply was YES -- but then the statement was to use a malloc call that provided NO alignment -- just a byte boundary as normal on a malloc.

Please explain.

Thanks

donghengqaz, 2018-11-02:
I don't know the DMA, but the sending payload must at DRAM, should not at IRAM. Otherwise a WIFI error will show and crash.

jhinklejr, 2018-11-02:
I understand - thanks.

Please close

closed 2018-11-04 by FayeY