espressif/ESP8266_RTOS_SDK #366: 编译问题

NickAABB, 2018-11-01:
mayn@DESKTOP-BR09NBL MINGW32 ~
# ls
esp  ESP8266_RTOS_SDK

mayn@DESKTOP-BR09NBL MINGW32 ~
# cd ESP8266_RTOS_SDK/

mayn@DESKTOP-BR09NBL MINGW32 ~/ESP8266_RTOS_SDK
# ls
components  docs  examples  helloWord  Kconfig  LICENSE  make  README.md  tools

mayn@DESKTOP-BR09NBL MINGW32 ~/ESP8266_RTOS_SDK
# cd helloWord/

mayn@DESKTOP-BR09NBL MINGW32 ~/ESP8266_RTOS_SDK/helloWord
# make menuconfig
make: *** 没有规则可制作目标“menuconfig”。 停止。

mayn@DESKTOP-BR09NBL MINGW32 ~/ESP8266_RTOS_SDK/helloWord
# make all
make: *** 没有规则可制作目标“all”。 停止。

mayn@DESKTOP-BR09NBL MINGW32 ~/ESP8266_RTOS_SDK/helloWord
#

donghengqaz, 2018-11-02:
你设置了 IDF_PATH 吗？

shiloye, 2018-11-05:
简单，你解压的时候没解压完全，
xtensa-lx106-elf
完全解压后
[image]
如果不是这个差不多大小 那就是这里少文件了

NickAABB, 2018-11-06:
>           请问，这个问题现在解决了吗？

[image]

export PATH="$PATH:/opt/xtensa-esp32-elf/bin"
export PATH="$PATH:/opt/xtensa-lx106-elf/bin"
export IDF_PATH="C:/msys32/home/mayn/ESP8266_RTOS_SDK"
这样设置的，环境变量，但是么有效果。

27-301, 2018-11-06:
export IDF_PATH="/home/mayn/ESP8266_RTOS_SDK" 试下。

NickAABB, 2018-11-15:
已经解决了。