espressif/ESP8266_RTOS_SDK #367: Memory Corruption with esp_aio_sendto()

jhinklejr, 2018-11-02:
I've modified my version of your heap (malloc and free) and have placed headers and trailers bracketing the  acquired buffer to detect memory over/under runs.

I am directly calling esp_aio_sendto() with data to be sent over one of the wifi interfaces (ESP Socket).

I acquire a buffer to hold the data being sent (malloc ) and assign as follows:

    aio.fd = Socket;
    aio.pbuf = (sbyte*)PP;
    aio.len = SZ;
    aio.cb = low_level_send_cb;
    aio.arg = PP;    // was p
    aio.ret = 0;

Socket is the acquired during call to tcpip_adapter_bind_netcard();
PP is buffer
SZ is the size of the buffer
aio.arg = PP;  .... saves the pointer so it can be freed in the callback.

In side low_level_send_cb() - I have verified that PP acquired from malloc above is the same as freed in low_level_send_cb();

Upon freeing PP (it was processed by esp_aio_sendto() - I detect that the buffer's header is incorrect indicating an memory corruption at the head of the buffer.

Please confirm that the buffer is not expected to be a lwip type buffer -- just a simple string buffer.

Any thoughts on the corruption?

I've tried make the buffer sent to esp_aio_sendto() 8bit aligned and 32bit aligned -- same results - corruption.

Thanks in advance for any insight

donghengqaz, 2018-11-02:
Yes it is, it is that. Wi-Fi don't dump a new buf, so add the WIFI packet header to the previous of the payload, so you should not using PP, but PP + bytes(wifi header length, you can find it at LwIP).

jhinklejr, 2018-11-02:
I should have figured there was a wifi header.

All is well.

Please close

closed 2018-11-04 by FayeY