espressif/ESP8266_RTOS_SDK #368: WIFI MAC Filter

jhinklejr, 2018-11-03:
Using tcpip_adapter_recv_cb() to get all the packets from the radio.

I'm receiving packets with destination MAC addresses not broadcast, and not those associated with the AP or STA interface.

Is there a filtering mechanism for weeding out all the packets I'n not interested in or must I do the filtering myself?

I ask because on Ethernet MAC, they usually have MAC filtering in hardware -- not sure if that capability exists on the ESP.

Thanks in adavance for your reply

Junhao-Espressif, 2019-01-15:
Hi @jhinklejr 

We can not set wifi mac filter when station is connect to AP by now.

Could you try with espnow or sniffer mode?

FayeY, 2019-04-17:
Closing due to lack of response, please feel free to reopen if this is still an issue.
Thanks.

closed 2019-04-17 by FayeY