espressif/ESP8266_RTOS_SDK #369: why it took two seconds before "scandone"

shiloye, 2018-11-05:
how do i spent less time to scande the wifi ?
i seems no working

i know my wifi   channle is 1,ssidi,password,and authmode
What else do I need to know? 
[image]

[image]

Junhao-Espressif, 2019-01-15:
Hi @shiloye 

Very good question!

Because it will scan all 13 channels by now. We will change it on next verison.

Thank you!

shiloye, 2019-01-25:
> Hi @shiloye
> 
> Very good question!
> 
> Because it will scan all 13 channels by now. We will change it on next verison.
> 
> Thank you!

i thingk that your answer is not about my question 
maybe u could try to using that 'fast scan'

Junhao-Espressif, 2019-04-10:
Hi @shiloye 

Please try again with latest VERSION.

Thank you!

FayeY, 2019-04-17:
Closing this,  please feel free to reopen if this is still an issue with the latest version.
Thanks.