espressif/ESP8266_RTOS_SDK #371: esp-idf V3.0 how to config the component.mk file ?

1XiaSheng, 2018-11-05:
[image]
[image]

1XiaSheng, 2018-11-05:
Great God, please see the above figure, what configuration is required for each component.mk file in the directory structure?

The correct guidelines are not available at https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html.

大神,请看上面的图,目录结构中的各个component.mk文件需要填入什么配置?

在https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html 中无法得到正确指引.

27-301, 2018-11-06:
for folder structure, you can refer to https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#example-project, you can rename app to components, for each component.mk in usr/hdw/cfg, you can refer to SDK/components/jsmn/component.mk.

1XiaSheng, 2018-11-07:
I hope to explain the specific example in the above picture, otherwise I still don't know how to add it.thanks
我希望是以上图中的具体的例子来说明,否则还是不知道怎么添加.谢谢

27-301, 2018-11-07:
you can upload your example here. But hope you can read the document and refer to components in SDK as example.

donghengqaz, 2018-11-07:
在你的 Makfile 里面添加:
```
EXTRA_COMPONENT_DIRS := $(shell pwd)/app
```
去掉 app 目录里面的 component.mk

1XiaSheng, 2018-11-07:
[图片]

1XiaSheng, 2018-11-07:
[image]
[image]

按这个结构配置成功.

jhinklejr, 2018-11-10:
1XiaSheng

Unless you are doing something advanced for the files specific, leave them empty.

An example of when or how to use them -- I was having a fault, and wanted to use a mix c and assembler listing to match the exception address to an exact line of code.

I added a single line, to that specific make file, which added extra CFLAG settings.  I removed it when I was done.

I've been doing a lot of advanced work and have yet, other than the example above, to make any modifications to the component make file.

They are there, in the default make process, to use if necessary.

Hope that helps

closed 2018-11-13 by FayeY