espressif/ESP8266_RTOS_SDK #382: I2s and DMA support missing/not working

THKDev, 2018-11-12:
## Environment

- Development Kit: ESP12-E
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    9d6c5608f13057972f4b208f446bdaac1f839111

## Problem Description
I2s support is missing. 

Are there any plans to bring I2s back into the SDK? I've try to get the [MP3 Decode](https://github.com/espressif/esp8266_mp3_decoder) example to run. Added missing headers and definitions.
But it seems that the I2s and DMA is disabled anywhere inside the SDK. No I2s interrupted is triggered. 
I2s and DMA are working on the same chip when i run an Arduino ESP8266 example.

Did i miss something?

THKDev, 2018-11-12:
Found the DMA problem. DMA memory has do be allocated by `heap_caps_malloc(size, MALLOC_CAP_DMA)` .

Now only some headers (i2s_reg.h and slc_register.h) and some defines are missing.

for instance:
```
#define ETS_SLC_INUM       1
#define ETS_SLC_INTR_ATTACH(func, arg)   _xt_isr_attach(ETS_SLC_INUM, (func), (void *)(arg))
#define ETS_SLC_INTR_ENABLE() _xt_isr_unmask(1 << ETS_SLC_INUM)
#define ETS_SLC_INTR_DISABLE() _xt_isr_mask(1 << ETS_SLC_INUM)

```

kmalloc8, 2018-11-27:
mark

zinahe, 2019-02-02:
Hi,

Can someone briefly explain how  DMA is now dependent on the capabilities-based heap allocation ? If I am not mistaken, it used to work with just **malloc()**. I'm trying to understand the reasoning behind heap allocation by capability. Is it to avoid memory fragmentation? or there is a physical limitation ?

Thanks,

Zinahe

berkutta, 2019-04-10:
Is there in the meantime a solution to use I2S in the new idf style sdk?

xiongyumail, 2019-04-11:
@berkutta 
Yes, I2S and SPI ram drivers will be available this month

FayeY, 2019-04-24:
Please have a try with the latest master, it has supported the i2s. Thanks.

closed 2019-05-20 by FayeY