espressif/ESP8266_RTOS_SDK #383: Sorry for late reply, as you've told I've tried it in my linux environment but still there is no effect after changing configTICK_RATE_HZ and recompiling. I've been waiting for months for the developers to make it configurable but still no fruitful outcome has come. Did I miss something here?

Mr-Techtron, 2018-11-13:
_Originally posted by @Mr-Techtron in https://github.com/espressif/ESP8266_RTOS_SDK/issues/176#issuecomment-391780759_

xiongyumail, 2018-11-15:
Please re-pull RTOS 2.x.x and compile freertos

Mr-Techtron, 2018-11-16:
I've re-pull RTOS 2.x.x and compile freertos but still there is no effect. FreeRTOS tick is still 10ms

Mr-Techtron, 2018-11-22:
can someone please guide me here?

donghengqaz, 2018-12-01:
You must change the macro "XT_TICK_DIVISOR", it will change the CPU tick cycle. config_TICK_RATE_HZ just affect the FreeRTOS not hardware.

Mr-Techtron, 2018-12-07:
@donghengqaz, I did what you said and recompile freertos but still, I can't change my CPU tick cycle. Have you tried this out by yourself? Is there something that I'm missing here?

FayeY, 2018-12-20:
Hi, sorry that RTOS 2.x.x does not support this feature, but RTOS 3.1 does. Would you like to have a try with the RTOS 3.1 branch? It is the new IDF style SDK that we will focus on.

FayeY, 2018-12-28:
Closing for now due to lack of response, please feel free to reopen if this is still an issue with the latest SDK. Thanks.

closed 2018-12-28 by FayeY