espressif/ESP8266_RTOS_SDK #385: Some functions in "driver/uart.c" has been deleted

wangym307, 2018-11-20:
INSTRUCTIONS
============
As we know, esp8266 use uart0 to print log defaultly.  
In old sdk(esp8266 sdk 1.0), we can use `os_install_putc1(uart1_write_char)`to make **uart1** print log, However, in newest idf-sdk (esp8266 sdk 3.0), `driver/uart.c` has been rewrited , and functions `uart1_write_char(char c), uart_tx_one_char(uint8 uart, uint8 TxChar)` were  deleted.   

OLD SDK CODE
============
```
LOCAL void
uart1_write_char(char c)
{
    if (c == '\n') {
        uart_tx_one_char(UART1, '\r');
        uart_tx_one_char(UART1, '\n');
    } else if (c == '\r') {
    } else {
        uart_tx_one_char(UART1, c);
    }
}

LOCAL void
uart0_write_char(char c)
{
    if (c == '\n') {
        uart_tx_one_char(UART0, '\r');
        uart_tx_one_char(UART0, '\n');
    } else if (c == '\r') {
    } else {
        uart_tx_one_char(UART0, c);
    }
}
```
Problem
============
How can I use uart1 to print log in esp-idf sdk?

xiongyumail, 2018-11-20:
Hi @wangym307 
Please change the UART0 in example to UART1.

wangym307, 2018-11-20:
sorry， @xiongyumail   
I don't know which example do you mean, can you describe it more detailedly?

xiongyumail, 2018-11-20:
https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/peripherals/uart_echo

wangym307, 2018-11-20:
@xiongyumail ,  
Thanks, but this example is used to set uart0 or uart1 as user uart, that means you can use uart0/uart1 to send serial data.   
What I need is `setting uart1 as system log output port`, not communicate with other serial port.    

just like below :
```
I (40) boot: SPI Speed      : 40MHz
I (43) boot: SPI Mode       : QIO
I (46) boot: SPI Flash Size : 1MB
I (48) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (55) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (65) boot:  2 factory          factory app      00 00 00010000 000e0000
I (69) boot:  3 tuya             Unknown data     01 aa 000f6000 0000a000
I (74) boot: End of partition table
I (77) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x992c8 (627400) map
I (220) esp_image: segment 1: paddr=0x000a92e0 vaddr=0x3ffe8000 size=0x00b20 (  2848) load
I (221) esp_image: segment 2: paddr=0x000a9e08 vaddr=0x3ffe8b20 size=0x0011c (   284) load
I (224) esp_image: segment 3: paddr=0x000a9f2c vaddr=0x40100000 size=0x07114 ( 28948) load
I (236) boot: Loaded app from partition at offset 0x10000
I (13) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (17) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
SDK version:v3.1-dev-347-g57c6215-dirty
...
...
```

xiongyumail, 2018-11-20:
@wangym307 
menuconfig can choose log print serial port.
menuconfig，Component config > ESP8266-specific > UART peripheral to use for console output (0-1)

wangym307, 2018-11-20:
@xiongyumail 
It solves, Thank you. 
I have another question: since log print port is defined by sdkconfig, and sdkconfig can only be modified before compiling,  does that mean we can't swap log print from uart0 to uart1 dynamically, after program running,  while  in esp8266 sdk 1.0, that is possible.

closed 2018-11-22 by wangym307