espressif/ESP8266_RTOS_SDK #386: MQTT重连会发送内存泄漏

ichenxing, 2018-11-22:
## Environment

- Development Kit: ESP8266_RTOS_SDK v2.1.0
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    f079158
- Development Env:     Make
- Operating System:     Ubuntu
- Power Supply:         USB

## Problem Description

当mqtt发布消息出现失败，断开重新连接时，内存会减少。[关于 mqtt demo 重连内存问题](https://github.com/espressif/ESP8266_RTOS_SDK/issues/225)

### Expected Behavior

### Actual Behavior

### Steps to repropduce

### Code to reproduce this issue

```
#include <stddef.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "mqtt/MQTTClient.h"

#include "esp_common.h"
#include "user_config.h"

#define MQTT_CLIENT_THREAD_NAME         "mqtt_client_thread"
#define MQTT_CLIENT_THREAD_STACK_WORDS  2048	//ret=1678 768
#define MQTT_CLIENT_THREAD_PRIO         8
#define MQTT_CLIENT_THREAD_BUf_SIZE		80

LOCAL xTaskHandle mqttc_client_handle;

static void messageArrived(MessageData* data)
{
    printf("Message arrived: %s\n", data->message->payload);
	
	memset(data->message->payload, 0, data->message->payloadlen);
}

static void mqtt_client_thread(void* pvParameters)
{
    printf("mqtt client thread starts\n");
	printf("sys: %d, task: %d.\r\n",system_get_free_heap_size(), uxTaskGetStackHighWaterMark(NULL));
	
    MQTTClient client;
    Network network;
    int rc = 0, count = 0;	
	unsigned char sendbuf[MQTT_CLIENT_THREAD_BUf_SIZE], readbuf[MQTT_CLIENT_THREAD_BUf_SIZE] = {0};
	
	MQTTPacket_connectData connectData = MQTTPacket_connectData_initializer;
	MQTTPacket_willOptions willOptions = MQTTPacket_willOptions_initializer;
	
	willOptions.topicName.cstring = "esp8266/sign";
	willOptions.message.cstring = "out";
	
	connectData.MQTTVersion = 3;
	connectData.clientID.cstring = "esp8266_#1";
	connectData.keepAliveInterval = 60;
	connectData.cleansession = true;
	connectData.willFlag = true;
	connectData.will = willOptions;
	
    pvParameters = 0;
	
    NetworkInit(&network);
    MQTTClientInit(&client, &network, 5000, sendbuf, MQTT_CLIENT_THREAD_BUf_SIZE, readbuf, MQTT_CLIENT_THREAD_BUf_SIZE);

    char* address = MQTT_BROKER;
	
	printf("sys: %d, task: %d.\r\n",system_get_free_heap_size(), uxTaskGetStackHighWaterMark(NULL));
	
	for(;;){
		count = 0;
		
		while (wifi_station_get_connect_status() != STATION_GOT_IP) {
			printf("get ip fail.\r\n");
			vTaskDelay(2000 / portTICK_RATE_MS);
		}
		
		if ((rc = NetworkConnect(&network, address, MQTT_PORT)) != 0) {
			printf("Return code from network connect is %d\n", rc);
			vTaskDelay(2000 / portTICK_RATE_MS);
			continue;
		} else{
			printf("Network Connect\n");
		}
		
		printf("sys: %d, task: %d.\r\n",system_get_free_heap_size(), uxTaskGetStackHighWaterMark(NULL));

#if defined(MQTT_TASK)
		if ((rc = MQTTStartTask(&client)) != pdPASS) {
			printf("Return code from start tasks is %d\n", rc);
		} else {
			printf("Use MQTTStartTask\n");
		}
#endif

		printf("sys: %d, task: %d.\r\n",system_get_free_heap_size(), uxTaskGetStackHighWaterMark(NULL));

		if ((rc = MQTTConnect(&client, &connectData)) != 0) {
			printf("Return code from MQTT connect is %d\n", rc);
			//MQTTDisconnect(&client);
			network.disconnect(&network);
			vTaskDelay(2000 / portTICK_RATE_MS);
			continue;
		} else {
			printf("MQTT Connected\n");
		}
		
		printf("sys: %d, task: %d.\r\n",system_get_free_heap_size(), uxTaskGetStackHighWaterMark(NULL));
		
		if ((rc = MQTTSubscribe(&client, "esp8266/sub", QOS2, messageArrived)) != 0) {
			printf("Return code from MQTT subscribe is %d\n", rc);
			//MQTTDisconnect(&client);
			network.disconnect(&network);
			vTaskDelay(2000 / portTICK_RATE_MS);
			continue;
		} else {
			printf("MQTT subscribe to topic \"esp8266/sub\"\n");
		}
		
		printf("sys: %d, task: %d.\r\n",system_get_free_heap_size(), uxTaskGetStackHighWaterMark(NULL));

		while (++count) {
			char payload[30];
			MQTTMessage message;
			message.qos = QOS2;
			message.retained = 1;
			message.payload = payload;
			sprintf(payload, "message number %d", count);
			message.payloadlen = strlen(payload);

			if ((rc = MQTTPublish(&client, "esp8266/pub", &message)) != 0) {
				printf("Return code from MQTT publish is %d\n", rc);
				break;
			} else {
				printf("MQTT publish topic \"esp8266/pub\", message number is %d\n", count);
			}

			vTaskDelay(2000 / portTICK_RATE_MS);  //send every 2 seconds
			
			printf("sys: %d, task: %d.\r\n",system_get_free_heap_size(), uxTaskGetStackHighWaterMark(NULL));
		}
		
		printf("MQTT disconnect.\r\n");
		//MQTTDisconnect(&client);
		network.disconnect(&network);
		vTaskDelay(2000 / portTICK_RATE_MS);
		printf("MQTT restart.\r\n");
		printf("sys: %d, task: %d.\r\n",system_get_free_heap_size(), uxTaskGetStackHighWaterMark(NULL));
	}

    printf("mqtt_client_thread going to be deleted\n");
    vTaskDelete(NULL);
    return;
}

void user_conn_init(void)
{
    int ret;
	
    ret = xTaskCreate(mqtt_client_thread,
                      MQTT_CLIENT_THREAD_NAME,
                      MQTT_CLIENT_THREAD_STACK_WORDS,
                      NULL,
                      MQTT_CLIENT_THREAD_PRIO,
                      &mqttc_client_handle);

    if (ret != pdPASS)  {
        printf("mqtt create client thread %s failed\n", MQTT_CLIENT_THREAD_NAME);
    }
}
```

## Debug Logs

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 30368, room 16 
tail 0
chksum 0x3a
load 0x3ffe8000, len 2008, room 8 
tail 0
chksum 0x26
load 0x3ffe87e0, len 896, room 8 
tail 8
chksum 0xd7
csum 0xd7
OS SDK ver: 2.0.0(7408579) compiled @ Sep 17 2018 13:41:04
phy ver: 1055_8, pp ver: 10.7

rf cal sector: 507
idle_task_hdl : 40107ad8,prio:0, stack:384
tim_task_hdl : 40107c20, prio:2,stack:512
SDK version:2.0.0(7408579) 71216

MQTT publish topic "esp8266/sub", message number is 1411
sys: 51968, task: 1666.
MQTT publish topic "esp8266/sub", message number is 1412
sys: 51968, task: 1666.
MQTT publish topic "esp8266/sub", message number is 1413
ip:192.168.1.130,mask:255.255.255.0,gw:192.168.1.1
sys: 51936, task: 1666.
MQTT publish topic "esp8266/sub", message number is 1414
sys: 51936, task: 1666.

MQTT publish topic "esp8266/sub", message number is 272
sys: 55600, task: 1670.
MQTT publish topic "esp8266/sub", message number is 273
sys: 55600, task: 1670.
MQTT publish topic "esp8266/sub", message number is 274
sys: 55600, task: 1670.
Return code from MQTT publish is -1
MQTT disconnect.
MQTT restart.
sys: 54816, task: 1670.
Return code from network connect is -1
Network Connect
sys: 57328, task: 1670.
Use MQTTStartTask
sys: 53704, task: 1670.
MQTT Connected
sys: 53568, task: 1670.
MQTT subscribe to topic "esp8266/sub"
sys: 53568, task: 1670.
MQTT publish topic "esp8266/sub", message number is 1
sys: 53568, task: 1670.
MQTT publish topic "esp8266/sub", message number is 2
sys: 53568, task: 1670.
MQTT publish topic "esp8266/sub", message number is 3
sys: 53568, task: 1670.
MQTT publish topic "esp8266/sub", message number is 4

MQTT publish topic "esp8266/sub", message number is 7
Network Connect
sys: 53568, task: 1670.
Use MQTTStartTask
sys: 49936, task: 1670.
MQTT Connected
sys: 49936, task: 1670.
MQTT subscribe to topic "esp8266/sub"
sys: 49936, task: 1670.
Return code from MQTT publish is -1
MQTT disconnect.
MQTT restart.
sys: 49152, task: 1670.
Network Connect
sys: 47720, task: 1670.
Use MQTTStartTask
sys: 44088, task: 1670.
MQTT Connected
sys: 44272, task: 1670.
MQTT subscribe to topic "esp8266/sub"
sys: 44272, task: 1670.
MQTT publish topic "esp8266/sub", message number is 1
sys: 46096, task: 1670.
MQTT publish topic "esp8266/sub", message number is 2

```

## Other items if possible

xuhongv, 2018-11-24:
@ichenxing 请问你的是哪个版本的sdk，我这边有个2.0版本的，经过测试稳定性好;
一个稳定、低耦合的适配乐鑫源码MQTT协议的esp8266库。 https://github.com/xuhongv/esp8266-xpMqtt

ichenxing, 2018-11-24:
@xuhongv release v2.1.0的sdk

FayeY, 2018-12-28:
你好，
我们最新重构了 [MQTT 示例](https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/protocols/mqtt)，建议使用最新版本的示例。
如果问题仍然存在，欢迎再与我们联系。
感谢对 ESP 的关注！

ustccw, 2019-01-02:
@xiongyumail `v3.1` 版本的 `SPI` 驱动开放了吗？

FayeY, 2019-01-23:
[V3.1 RC1](https://github.com/espressif/ESP8266_RTOS_SDK/issues/402) 已提供 SPI 驱动，请知悉，谢谢！

ichenxing, 2019-01-23:
@FayeY 以测试新版本的HSPI，还有点问题

FayeY, 2019-01-23:
@ichenxing  HSPI 的问题烦请另提一个 issue，我们将指派 HSPI 的工程师协助。
想请问，这个 MQTT 的 issue 解决了吗？新版本我们重构了 MQTT，一般都会推荐客户使用新版本。
请知悉。谢谢！

ichenxing, 2019-01-23:
@FayeY新版本的mqtt还没有进行长时间的测试；关于hspi的问题已经提了一个issue；

FayeY, 2019-02-21:
暂时关闭此问题，如果新版 MQTT 仍然有问题，欢迎 reopen。谢谢！

closed 2019-02-21 by FayeY

sxyandapp, 2021-07-17:
rtos 2.0.0版本，将NetworkInit()和MQTTClientInit放到循环体外边时，内存泄漏问题消失
2021年7月17日