espressif/ESP8266_RTOS_SDK #387: smart config 配置问题

zhubinghan, 2018-11-22:
我用乐鑫新发布的 esp8266 sdk3.0，用自带的例子，但是smartconfig成功率很低。10次里不一定会有一次成功。手机是安卓7.0的小米5手机，esptouch版本是v0.3.7.0  。不知道有没有办法提高配置成功率。
----------------------------- Delete above -----------------------------

## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:      [Arduino IDE|Make|Eclipse|other]
- Operating System:     [Windows|Ubuntu|MacOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

hvritual, 2018-12-22:
😳 一样遇上的这个问题_ 我用的是ios12.1  EspTouch 0.3.7 现在在抓手机发送的udp报文

Werner42, 2018-12-22:
上面用iOS的。我之前试的时候也是，基本不会成功。但我看到前两天espressif esptouch这个app更新了。写着修复了iOS12下的wifi info问题。我还没试过，你可以试一下。

hvritual, 2018-12-22:
> 上面用iOS的。我之前试的时候也是，基本不会成功。但我看到前两天espressif esptouch这个app更新了。写着修复了iOS12下的wifi info问题。我还没试过，你可以试一下。

我这边就是使用最新版本的ios 还是不能正常工作 现在准备是用Android 来进行测试一下

ustccw, 2019-01-02:
可以基于最近的 master 测试一下, 看看问题是否还存在.

Werner42, 2019-01-04:
> 可以基于最近的 master 测试一下, 看看问题是否还存在.
我刚才pull一下新的，用的ios 下的 espressif esptouch，现在没有卡死可以配置成功了。

hvritual, 2019-01-04:
最新的固件可以正常使用smartconfig了 配置不是esptouch的原因 是固件的问题 后面8266的固件重新升级了

27-301, 2019-01-04:
非常感谢 @Werner42 @hvritual 两位同步的信息，最近对 smartconfig 做了较大的优化。

27-301, 2019-01-08:
谢谢各位的反馈，先关闭该问题，如需要新问题，请再提出。

closed 2019-01-08 by 27-301