espressif/ESP8266_RTOS_SDK #388: Meaning of esp_aio_sendto() CB error

jhinklejr, 2018-11-23:
esp_aio_sendto() return an error code via it's CB function -- low_level_send_cb().

The CB function return aio->ret as an error code.

What does a return of 2 mean?

What does a return of 4 mean?

Are these error codes documented someplace?

Thanks

jhinklejr, 2018-11-24:
I think I may have found the enum for the replies - esp_wifi_types.h

What does TX_STATUS_SRC_EXCEED mean?

What does TX_STATUS_LRC_EXCEED mean?

jhinklejr, 2018-11-24:
Here is a line of code from LWIP -- 
        LWIP_DEBUGF(PBUF_CACHE_DEBUG, ("Send packet fail: result:%d, LRC:%d, SRC:%d, RATE:%d",
                                       status->wifi_tx_result, status->wifi_tx_lrc, status->wifi_tx_src, status->wifi_tx_rate));

What is wifi_tx_lrc?

What is wifi_tx_src?

Thanks

donghengqaz, 2018-12-01:
@Junhao-Espressif

jhinklejr, 2018-12-01:
Hi

 

I don’t understand your reply – just a user name.

 

I had asked what These variables were:

 

What is wifi_tx_lrc?

What is wifi_tx_src?

 

They are part of the argument returned to the Transmit CBV function:

 

esp_aio_sendto() return an error code via it's CB function -- low_level_send_cb().

 

Thanks

donghengqaz, 2018-12-03:
Hi, I am sorry that i don't master at the field of Wi-Fi, but JunHao he is the one.

Junhao-Espressif, 2018-12-18:
Hi @jhinklejr ,

src means the number of times rts was send before AP send cts to us.

lrc means the number of times unicast data was send before AP send ack to us.

closed 2018-12-18 by Junhao-Espressif