espressif/ESP8266_RTOS_SDK #394: esp-idf框架8266-2Mflash下ota1分区的offset必须比ota0大0x100000？

wangym307, 2018-11-26:
我使用2MB的flash，patitions_table是custom模式，自定义的分区表为
```
# Name,   Type, SubType, Offset,   Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x4000
otadata,  data, ota,     0xd000,  0x2000
phy_init, data, phy,     0xf000,  0x1000
ota_0,    0,    ota_0,   0x10000, 0xd0000
ota_1,    0,    ota_1,   0xe0000, 0xd0000
my_data, data, 0xAA,     0x1b0000,0x28000,
```
然后发现ota完成后固件无法从ota_1的地址0xe0000启动。
而如果我将ota_1的起始地址改为0x110000就可以ota后正常启动。  
查看文档 https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/docs/en/api-guides/partition-tables.rst 
里面有这样一段描述：

> Offset & Size
Please note that the app partition must fall in only one integrated partition of 1M. Otherwise, the application crashes.
The starting address of firmware is configured to 0x10000 by default. If you want to change the starting address of firmware, please:
Configure the value in menu -> partition table -> select "Custom partition table CSV" -> (0x10000) Factory app partition offset;
Configure the ota_1 offset in the CSV file of partition table to the value , and ota_2 offset to the mirror value (ota_2 = ota_1 + 0x100000).
Please enter an aligned offset. Otherwise, the tool will return errors.
Don't leave it blank, because, in this case, the tool will automatically align the app partition, which may cause app partition overlaps. That said, the app partition falls in more than one integrated partitions of 1M.
Sizes and offsets can be specified as decimal numbers, hex numbers with the prefix 0x, or size multipliers K or M (1024 and 1024*1024 bytes).

### 两个疑问
#### 一 、用户区固件在flash中不能横跨0x100000地址？
第一句话“Please note that the app partition must fall in only one integrated partition of 1M”是否意味着一个app用户区bin文件不能跨越1M 地址，也就是ota_0的结束地址不能超过0x100000，这一点似乎与
ESP8266_RTOS_SDK/components/bootloader_support/src/bootloader_utility.c的代码一致
```
    if (image_data->start_addr < ESP_CACHE1_ADDR_MAX
        && image_data->start_addr + image_data->image_len < ESP_CACHE1_ADDR_MAX) { 
        map = 0;
    } else if (image_data->start_addr >= ESP_CACHE1_ADDR_MAX
               && image_data->start_addr < ESP_CACHE2_ADDR_MAX
               && image_data->start_addr + image_data->image_len < ESP_CACHE2_ADDR_MAX) {
        map = 1;
    } else {
        ESP_LOGE(TAG, "ERROR: app bin error, start_addr %x image_len %d\n", image_data->start_addr, image_data->image_len);
        /* Blocking here to let user judge. */
        while (1);
    }

```
#### 二  、ota_2 = ota_1 + 0x100000？
如果ota_2必须比ota_1大0x100000吗？有没有什么方式可以修改，因为我的固件大小只有600-800KB，如果ota_2必须放在比ota_1大1MB（0x100000）的位置，那么从ota_0的结束位置到0x110000之间将会空出一段200KB左右的flash空间，从ota_1的结束位置到0x200000之间也会空出200KB左右的flash空间。目前我的存储管理是面向一整块flash空间的（分区表中的`my_data`），如果选择其中一块来存储数据，那么另一块flash空间就浪费掉了。

wangym307, 2018-11-28:
@filzek Thank you,
As you said, I know that a partition can't lies in 2 sets of 1MB. So, if I use 2MB flash, the ota_1' offset must be larger than 0x100000. 
However, when I change the start address of ota_1 to 0x100000 or 0x120000, after ota, the program still can't work from  0x100000.
#### What I changed:
```
1.make menuconfig->Partition Table ->Custom partition table CSV, 
   CSV file rename to my_partitions.csv.
2.rebuild custom my_partitions.bin , and flash my_partitions.bin to 0x8000
```
#### Debug Log
```
load 0x40100000, len 7252, room 16 
tail 4
chksum 0x36
load 0x3ffe8408, len 24, room 4 
tail 4
chksum 0xbf
load 0x3ffe8420, len 3500, room 4 
tail 8
chksum 0x25
csum 0x25�I (36) boot: ESP-IDF cafd0c1-dirty 2nd stage bootloader
I (36) boot: compile time 11:34:02
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : QIO
I (43) boot: SPI Flash Size : 2MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (65) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  3 ota_0            OTA app          00 10 00010000 000d0000
I (88) boot:  4 ota_1            OTA app          00 11 00100000 000d0000
I (95) boot:  5 my_data          Unknown data     01 aa 001d0000 00028000
I (102) boot: End of partition table
I (107) esp_image: segment 0: paddr=0x00100010 vaddr=0x40210010 size=0x9b848 (637000) map
I (254) esp_image: segment 1: paddr=0x0019b860 vaddr=0x3ffe8000 size=0x00b54 (  2900) load
I (255) esp_image: segment 2: paddr=0x0019c3bc vaddr=0x3ffe8b54 size=0x0011c (   284) load
I (262) esp_image: segment 3: paddr=0x0019c4e0 vaddr=0x40100000 size=0x07118 ( 28952) load
I (276) boot: Loaded app from partition at offset 0x100000
```

#### Question:
Only when I set ota_1's start_address to 0x110000, after ota, program can start normally.
Did I miss some configuration? Or I need to change something in code?

donghengqaz, 2018-12-01:
一个 APP 不能横跨 1MB 分界处，但是可以坐落在：第一个1M，第二个1M 里面。

wangym307, 2018-12-01:
@donghengqaz ，  
谢谢，从1MB版本的partitions_two_ota.1MB.csv示例中也得到了确认。
现在还有一个问题是：如果要修改ota1的起始地址，除了改烧录的partition.csv以及生成的bin文件，还需要修改什么东西吗？
因为我测试发现，如果把ota1的起始地址改为0x110000以外的值，程序ota完成之后都不能从ota1启动。

filzek, 2018-12-02:
@wangym307 I did some more testing and you are right, someplaces in between 1MB will not work even if the laid in 1MB block. Also there is a problem setting more the 2 partition of 1MB it doesnt not allow creation of any more partition beyond 2MB, I did not found why yet.

donghengqaz, 2018-12-03:
@wangym307 由于 ESP8266 没有 MMU 所以不能做 flash 映射，使得现在的版本做了诸多限制，防止用户出现各种异常，最直接的结果就是 OTA0 和 OTA1 必须 offset 相同，这是便于用户使用。

如果过你要强制修改 OTA1 的地址，可以在编译 OTA1 的时候使用命令 `make APP_OFFSET=xxxxx APP_SIZE=yyy` 其中 `xxxxx` 为 OTA1 便宜地址，`yyy`  为 partition 大小。

closed 2018-12-20 by FayeY

wangym307, 2019-01-17:
@donghengqaz 
你好，为了使得ota1分区的地址能从默认的0x110000改为0x100000。我按照你的说法：
在生成ota1用户区固件的时候为make指定参数APP_OFFSET=0。
ota的时候我从代码日志看出来固件确实是写到0x100000的地址上了（这个应该是由partitions.bin决定的，与APP_OFFSET无关），但是ota之后，固件仍然无法从0x100000加载。
从你们的代码来看`APP_OFFSET ?= $(CONFIG_APP1_OFFSET)`，如果没有指定的话，APP_OFFSET默认值是等于CONFIG_APP1_OFFSET，而CONFIG_APP1_OFFSET=0x10000，并且APP_OFFSET只影响了`make flash`时的烧写地址参数，并不会影响bootloader代码，是否bootloader还没有完全适配OTA1区OFFSET的变化？

日志：
```

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 7324, room 16 
tail 12
chksum 0xd9
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x50
ho 0 tail 12 room 4
load 0x3ffe8420, len 3564, room 12 
tail 0
chksum 0x50
csum 0x5032mI (65) boot: ESP-IDF common_test_181129-19-g615d717-dirty 2nd stage bootloader
I (65) boot: compile time 09:45:50
I (75) qio_mode: Enabling default flash chip QIO
I (76) boot: SPI Speed      : 40MHz
I (82) boot: SPI Mode       : QIO
I (88) boot: SPI Flash Size : 2MB
I (94) boot: Partition Table:
I (100) boot: ## Label            Usage          Type ST Offset   Length
I (111) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (123) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (134) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (146) boot:  3 ota_0            OTA app          00 10 00010000 000c0000
I (158) boot:  4 ota_1            OTA app          00 11 00100000 000c0000
I (169) boot:  5 custom             Unknown data     01 aa 000d0000 00028000
I (181) boot: End of partition table
I (187) esp_image: segment 0: paddr=0x00100010 vaddr=0x40210010 size=0x96388 (615304) map
I (408) esp_image: segment 1: paddr=0x001963a0 vaddr=0x3ffe8000 size=0x00b74 (  2932) load
I (409) esp_image: segment 2: paddr=0x00196f1c vaddr=0x3ffe8b74 size=0x0011c (   284) load
I (419) esp_image: segment 3: paddr=0x00197040 vaddr=0x40100000 size=0x06a24 ( 27172) load
I (441) boot: Loaded app from partition at offset 0x100000
```

wangym307, 2019-01-17:
@donghengqaz 我刚才又重试了一次，指定OFFSET，在编译之前make clean一下，发现OFFSET是可以生效的，现在已经ota成功，固件可以从0x100000加载。但是这个也带来另外一个问题：
ota0和ota1的升级固件不能共用一个：ota0的OFFSET是0x10000， ota1的OFFSET是0x0。
这样对我们来说是很不便使用的，请问是否有其他的解决办法，让ota0和ota1的固件可以相同？