espressif/ESP8266_RTOS_SDK #398: ESP8266 Transmit Power Level - confusion

jhinklejr, 2018-11-28:
I'm having a problem with the ESP8266 is APSTA mode to have the STA connect with an AP with an rssi of -65.

I realize the rssi is an indication of the AP's transmit power level where our (my STA) transmit power level will determine if the AP can/will connect.

Your documentation of transmit power level is very confusing.

Your documentation states that default transmit power is level0 or 78-128.  It states that is the default level.

I've run a test to see if I could correlate the two functions:
esp_wifi_set_max_tx_power
esp_wifi_get_max_tx_power

I have stepped setting max power level from 0 to 120 in steps of 10 using esp_wifi_set_max_tx_power()
I delay 25 msec  to give you time to stabilize
I then read power level via esp_wifi_get_max_tx_power() which ALWAYS returns the value of 20.

These calls all occur after I have executed esp_wifi_start();

Something is wrong since esp_wifi_get_max_tx_power() always returns 20 no matter what value (from 0 to 120) is set via esp_wifi_set_max_tx_power();

Just to clarify -- calls to both functions always return 0 - no error.

Can you explain what is happening?

How can I ensure I have maximum transmit power level set?  

Why are the arguments and return value associated with the set/get transmit power level so different and confusing?

Here is the test code:

	int8_t Power = 0;
	int8_t P1;
	int r1, r;

	for(Power = 0; Power <= 110; Power += 10)
	{
		r1 = esp_wifi_set_max_tx_power(Power);

		vTaskDelay(5);

		r = esp_wifi_get_max_tx_power(&P1);
		My_printf("\n\nPower Set %d Return %d Power Get Return %d  Value %d\n\n", Power, r1, r, P1);

		vTaskDelay(5);

	}

The following is from the header file on these two functions -- very confusing.

Thanks in advance for any insights and replies.

/**
  * @brief     Set maximum WiFi transmiting power
  *
  * @attention WiFi transmiting power is divided to six levels in phy init data.
  *            Level0 represents highest transmiting power and level5 represents lowest
  *            transmiting power. Packets of different rates are transmitted in
  *            different powers according to the configuration in phy init data.
  *            This API only sets maximum WiFi transmiting power. If this API is called,
  *            the transmiting power of every packet will be less than or equal to the
  *            value set by this API. If this API is not called, the value of maximum
  *            transmitting power set in phy_init_data.bin or menuconfig (depend on
  *            whether to use phy init data in partition or not) will be used. Default
  *            value is level0. Values passed in power are mapped to transmit power
  *            levels as follows:
  *            - [78, 127]: level0
  *            - [76, 77]: level1
  *            - [74, 75]: level2
  *            - [68, 73]: level3
  *            - [60, 67]: level4
  *            - [52, 59]: level5
  *            - [44, 51]: level5 - 2dBm
  *            - [34, 43]: level5 - 4.5dBm
  *            - [28, 33]: level5 - 6dBm
  *            - [20, 27]: level5 - 8dBm
  *            - [8, 19]: level5 - 11dBm
  *            - [-128, 7]: level5 - 14dBm
  *
  * @param     power  Maximum WiFi transmiting power.
  *
  * @return
  *    - ESP_OK: succeed
  *    - ESP_ERR_WIFI_NOT_INIT: WiFi is not initialized by esp_wifi_init
  *    - ESP_ERR_WIFI_NOT_START: WiFi is not started by esp_wifi_start
  */
esp_err_t esp_wifi_set_max_tx_power(int8_t power);

/**
  * @brief     Get maximum WiFi transmiting power
  *
  * @attention This API gets maximum WiFi transmiting power. Values got
  *            from power are mapped to transmit power levels as follows:
  *            - 78: 19.5dBm
  *            - 76: 19dBm
  *            - 74: 18.5dBm
  *            - 68: 17dBm
  *            - 60: 15dBm
  *            - 52: 13dBm
  *            - 44: 11dBm
  *            - 34: 8.5dBm
  *            - 28: 7dBm
  *            - 20: 5dBm
  *            - 8:  2dBm
  *            - -4: -1dBm
  *
  * @param     power  Maximum WiFi transmiting power.
  *
  * @return
  *    - ESP_OK: succeed
  *    - ESP_ERR_WIFI_NOT_INIT: WiFi is not initialized by esp_wifi_init
  *    - ESP_ERR_WIFI_NOT_START: WiFi is not started by esp_wifi_start
  *    - ESP_ERR_INVALID_ARG: invalid argument
  */
esp_err_t esp_wifi_get_max_tx_power(int8_t *power);

FayeY, 2018-12-28:
Could you have a try with our latest master? We think that this issue should have been solved recently. Thanks.

FayeY, 2019-01-09:
Closing due to lack of response, please feel free to reopen if this is still an issue. Thanks.

closed 2019-01-09 by FayeY