espressif/ESP8266_RTOS_SDK #401: Commercial product with ESP8266 - fail to connect

jhinklejr, 2018-12-02:
I've got about 50 units it the field and customers are reporting they can't connect to their access points.

Data I am collecting indicates that event SYSTEM_EVENT_STA_DISCONNECTED is returning error 201 about 20% of the time and 2 the other 80%.

Some of the errors are being returned when the AP is less that 5 feet away.

I can't find good information on errors 201 and 2.  Are they Espressif's error numbers or are they being returned from the AP?

I'm trying to understand what is under my control that I can do to address this issue - it is affecting a lot of my customers.

Does phy_init have any impact?

Thanks

filzek, 2018-12-02:
maybe you compiled your software with 40mhz besides 26mhz, it does a lot of problem

jhinklejr, 2018-12-03:
Please explain.

It was set at 40.  I was unaware the spi speed had an impact.

Can you explain the association between the two.

Thanks

jhinklejr, 2018-12-03:
My ESP8266 WROOM is embedded on a larger board and the CPU on that board does the flash programming.

I set location 0x00003 to 0x60 which says spi speed in 40mhz.

The only place I find a crystal selection 40 vs 26 is in the "serial flasher config" section and it speaks to SPI speed.

Is the frequency selection also used to identify crystal selection elsewhere -- as I investigating this, I have noticed post talking about location 0x0000030 being used: 0 for 40mhz and 1 for 26mhz .... if that is so, how does that selection get set.

Sorry for all the tech questions, but I can't use the flash tool as stated above so I need to implement all the correct values myself.

Thanks

jhinklejr, 2018-12-03:
Are you speaking of 40mhz vs 26mhz in ESP8266_RF_init.xls?

If so, I'm not sure how to check

jhinklejr, 2018-12-03:
Forget the rabbit hole venture - I've found where the 40/26 mhz crystal is stated and I'm using 26mhz.

Thanks filzek for the comment - it took me on a several hour hunt for the information - I eventually found it -- learned some things along the way - again - thanks for the comment.

So -- I'm back to my original question"

I've got about 50 units it the field and customers are reporting they can't connect to their access points.

Data I am collecting indicates that event SYSTEM_EVENT_STA_DISCONNECTED is returning error 201 about 20% of the time and 2 the other 80%.

Some of the errors are being returned when the AP is less that 5 feet away.

I can't find good information on errors 201 and 2. Are they Espressif's error numbers or are they being returned from the AP?

I'm trying to understand what is under my control that I can do to address this issue - it is affecting a lot of my customers.

Does phy_init have any impact - other than crystal frequency?

FayeY, 2018-12-28:
Could you have a try with the latest master? This issue may have been solved. Please feel free to let us know, if it is not. Thanks.

closed 2019-01-09 by FayeY