espressif/ESP8266_RTOS_SDK #402: 什么时候支持hspi挂载flash？

Anerikamao, 2018-12-03:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at bbs.espressif.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:      [Arduino IDE|Make|Eclipse|other]
- Operating System:     [Windows|Ubuntu|MacOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

xiongyumail, 2018-12-03:
会先给到hspi驱动，可以在这上做对应flash的适配。

Anerikamao, 2018-12-03:
> 会先给到hspi驱动，可以在这上做对应flash的适配。

我使用v3.0sdk对ESP-WROOM-S2进行开发，ESP-WROOM-S2是使用hspi挂载flash的，使用esptool无法下载，出现如下log，翻查hanlde_spi_attach函数，有写ESP8266 doesn'nt yet support SPI on HPSI
现在有解决方案？

lin@ubuntu:~/esp/ESP8266_RTOS_SDK/examples/wifi/simple_wifi$ make flash
Flashing binaries to serial port /dev/ttyS1 (app at offset 0x10000)...
esptool.py v2.4.0
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:9c:b8:76
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 10688 bytes to 7110...

A fatal error occurred: Timed out waiting for packet content
/home/lin/esp/ESP8266_RTOS_SDK/components/esptool_py/Makefile.projbuild:68: recipe for target 'flash' failed
make: *** [flash] Error 2
linguangmao@ubuntu:~/esp/ESP8266_RTOS_SDK/examples/wifi/simple_wifi$

donghengqaz, 2018-12-03:
你可以检查下的串口是否正确，一般 ubuntu 下面 USB 转 串口设备的名字不会是 ttyS1 ，一般都是 ttyUSB(X) ，X=0,1,2....，什么的。

Anerikamao, 2018-12-03:
> 你可以检查下的串口是否正确，一般 ubuntu 下面 USB 转 串口设备的名字不会是 ttyS1 ，一般都是 ttyUSB(X) ，X=0,1,2....，什么的。

使用minicom检查过串口，是正常的。用esptool的命令读取芯片固化的信息也是ok的，但是进行flash相关的操作的命令就失败了，应该hspi挂载flash有关

27-301, 2018-12-27:
S2 模组上，SPI 是挂在 HSPI 上的。当前 SDK 和 esptool 都还没有能支持 S2 。