espressif/ESP8266_RTOS_SDK #406: Spiffs save/create files always return error

grays820, 2018-12-04:
void spiffs_fs1_init(void)
{
    struct esp_spiffs_config config;
    config.phys_size = FS1_FLASH_SIZE;
    config.phys_addr = FS1_FLASH_ADDR;
    config.phys_erase_block = SECTOR_SIZE;
    config.log_block_size = LOG_BLOCK;
    config.log_page_size = LOG_PAGE;
    config.fd_buf_size = FD_BUF_SIZE * 2;
    config.cache_buf_size = CACHE_BUF_SIZE;
    int x = esp_spiffs_init(&config);
    printf("init fs:%d \n",x);
}
 
void test_write()
{
    char *buf = "hello world";
    int pfd = open("myfile", O_TRUNC | O_CREAT | O_ACCMODE, S_IRUSR | S_IWUSR);
    if (pfd <= 3)
    {
        printf("open file error %d\n",pfd);
    }
    int write_byte = write(pfd, buf, strlen(buf));
    if (write_byte <= 0)
    {
        printf("write file error %d\n",write_byte);
    }
    close(pfd);
}

void test_read()
{
    char out[20] ={0};
    int pfd =open("myfile", O_RDWR);
    if (read(pfd, out, 20) < 0)
        printf("read errno \n");
    close(pfd);
    printf("--> %s <--\n", out);
}
void app_main(void)
{
    spiffs_fs1_init();
    test_write();
    test_read();

}

monitor return 

open file error 0
write file error -1
read errno
-->  <--

Can anyone tell me to save the file like this?

grays820, 2018-12-04:
smaple code copy from esp8266_rtos_sdk_programming_guide_cn.pdf

ustccw, 2019-01-02:
@all spiffs is not supported  currently. You could try `nvs` to save your configurations. refer to [nvs.h](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/nvs_flash/include/nvs.h)

filzek, 2019-01-02:
Spiff is supported and works perfect

WallaceWilliam, 2019-03-20:
hiperiondev
in res = esp_spiffs_init(&config);
set variable static fs.
but static fs from esp_spiffs.c is not static fs from this code
so SPIFFS_format(&fs) return -10024 SPIFFS_ERR_NOT_CONFIGURED
so you need 
1) in esp_spiffs.c declare fs not static
2) declare your variable fs as extern
3) possibly in esp_spiffs.h declare fs

FayeY, 2019-05-20:
SPIFFS is also added in the latest master, please have a try. Thanks.

closed 2019-05-20 by FayeY